microsoft/Azure-Kinect-Sensor-SDK #1164: Saving pointcloud as .ply with normals

philianeles, 2020-04-08:
Hi, I am using the Azure-Kinect-Sensor-SDK `K4A.sln `--> `transformation_example` to create the .ply sample.

I seem to have zero length normals, or normals with infinity/undefined coefficients.
Is there a way to retrieve and save point cloud (.ply for instance) and have the normals as well?

Thanks!

tesych, 2020-04-13:
@philianeles thank you for your question! Can you share more details?

philianeles, 2020-04-14:
Sure!
I seem to have one .ply file produced as a result of running the `transformation_example `.
Which can be found here: https://www.dropbox.com/s/4y64x38wuss29qw/color_to_depth.ply?dl=0 
If you check it contains x, y, z and the rgb values.
But I was wondering there should be a way to retrieve the normals too? (nx, ny, nz?)

Actually what I am looking for is to get the point cloud steamed and saved (could be .ply, .off, etc.) that has non-zero nx, ny, nz values.
Is it anyhow possible?

rabbitdaxi, 2020-04-21:
@philianeles thank you for the question. The transformation_example does not compute normal, it only compute point cloud (the example is just using ply as a simple format to allow user visualize the point cloud in MeshLab). To compute the normal from a depth point cloud, there are many ways to do that, naïve way is estimating normal from neighbors but need to be careful with the depth discontinuity area. Or you can read some more papers with advanced way. Feel free to contribute to the example with normal computation :)

closed 2020-07-11 by qm13

Alilfd, 2020-08-17:
> Hi, I am using the Azure-Kinect-Sensor-SDK `K4A.sln `--> `transformation_example` to create a .ply sample.
> 
> I seem to have zero length normals, or normals with infinity/undefined coefficients.
> Is there a way to retrieve and save point cloud (.ply) and have the normals stored in there as well?
> 
> Thanks!

Hi,
I have just a question on how can I generate point clouds from mkv file. I would appreciate it if you direct me. 

Thanks,

fatihoezdemir, 2020-09-10:
@Alilfd  if you used the azure kinect and recorded it in a .mkv file, just run the transformation_example from the Azure Kinect SDK

Lilibel87, 2020-11-27:
@fatihoezdemir hi, I have the mkv and I am just unsure where to run the transformatoin_example, from the prompt command as the recording? I need to register a sequence of .ply from the video. Is that possible as well? Thank you!

fatihoezdemir, 2020-11-27:
For Ubuntu: If you have the Azure SDK, then go to the build ordner from your terminal and run ./transformation_example playback <filename.mkv> [timestamp (ms)] [output_file]
I have installed the azure sdk from source but you should find it somewhere there

e.g. ./transformation_example playback test.mkv 20 test.ply
Here is a guiding link on how to use the programhttps://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/transformation

Lilibel87, 2020-11-29:
> Hi, I am using the Azure-Kinect-Sensor-SDK `K4A.sln `--> `transformation_example` to create a .ply sample.
>
> I seem to have zero length normals, or normals with infinity/undefined coefficients.
> Is [...]

Hello, I have extracted the mkv files using the - windows promt, but is not working with the ply. Where should I compile the transformation examples? Thank you!