microsoft/Azure-Kinect-Sensor-SDK #1165: Viewer and Recorder Errors.

sechternacht, 2020-04-08:
Hello,

I am just getting started working with the azure kinect.  Our ultimate goal is to use this the azure kinect for facial mapping in patients with facial paralysis.  However, I'm encountering a couple of errors that I can't seem to find a solution to.

1. When I go to open my device in the k4aviewer, I receive the error: "Failed to read color parameter: auto_exposure_priority!"  Although this error occurs, I am able to dismiss it and still view the three camera modes in realtime. 
2. Unsure if this is related to the auto_exposure_priority error, but when I attempt to record with the device using the following command:

"C:\Program Files\Azure Kinect SDK\tools>k4arecorder.exe output.mkv"

I receive the following error: 

"Device serial number: 000866194412
Device version: Rel; C: 1.6.102; D: 1.6.75[6109.7]; A: 1.6.14
Device started
[2020-04-08 15:22:44.560] [error] [t=7832] [K4aRecordApi] Unable to open file 'output.mkv': ios_base::failbit set: iostream stream error
Unable to create recording file: output.mkv"

I am currently using the following:
- OS with Version: Windows 10 Pro v1909
- SDK Version: 1.4.0
- Firmware Version:
     - RGB camera firmware:       1.6.102
     - Depth camera firmware:    1.6.75
     - Depth config file:               6109.7
     - Audio firmware:                 1.6.14

Any help with these issues would be greatly appreciated!

wes-b, 2020-04-08:
It seems like you might be mixing different versions of K4aViewer with K4a.dll. K4A_COLOR_CONTROL_AUTO_EXPOSURE_PRIORITY was depricated a couple releases back. 

Please open the K4aView, turn on logging, level to info and then open the sensor. Please share the log.

For your k4arecorder issue, it looks like an access denied error. Ensure you have permission to create a file there.

sechternacht, 2020-04-16:
Wes-b, thanks for your reply.  I was mixing different versions of k4aviewer with k4a.dll.  I have sorted this out and no longer have issues with the viewer.  However, for the k4arecorder issue I am now receiving a different error message (see below).  Please advise.  Thanks in advance.

C:\Program Files\Azure Kinect SDK v1.4.0\tools>k4arecorder.exe output.mkv
Device serial number: 000866194412
Device version: Rel; C: 1.6.102; D: 1.6.75[6109.7]; A: 1.6.14
[2020-04-15 21:31:33.389] [error] [t=3968] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\mfcamerareader.cpp (840): SetCameraControl(). K4A_COLOR_CONTROL_MODE_AUTO is not supported for 8 color control
[2020-04-15 21:31:33.390] [error] [t=3968] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (1019): color_set_control(device->color, command, mode, value) returned failure in k4a_device_set_color_control()
Runtime error: k4a_device_set_color_control() for auto gain failed
Device started
[2020-04-15 21:31:33.622] [error] [t=3968] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\record\sdk\record.cpp (40): k4a_record_create(). Unable to open file 'output.mkv': ios_base::failbit set: iostream stream error
Unable to create recording file: output.mkv

wes-b, 2020-04-16:
Hmm, the error about auto mode is a bug, so attached is the PR. As for the issue opening output.mkv, can you ensure you have the ability to create a file in the location you are asking the tool to create it? Also make sure your disk has sufficient space.

jackyxu, 2020-04-18:
Camera failed time out error for azure kinect
What should I do when I launch the "k4aviewer" SDK with "Azure Kinect SDK v1.4.0" and an error message of "camera failed time out" appears?

jackyxu, 2020-04-18:
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4435188 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4468533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4501855 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4535200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4568533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4601866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4635177 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4668533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4701855 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4735200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4768522 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4801866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4835188 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4868533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4901866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4935200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   4968533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5001866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5035200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5068533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5101866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5135200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5168533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5201866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5235200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5268533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5301866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5335200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5368533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5401866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5435200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5468533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5501866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5535200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5568533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5601866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5635200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5668533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5701866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5735200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5768533 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5801866 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5835200 type:Depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   5868533 type:Depth
[ error    ] : OnFlush(). Failed to deselect stream for stop: 0x800706bf

wes-b, 2020-04-20:
> [ error ] : OnFlush(). Failed to deselect stream for stop: 0x800706bf

#381

closed 2020-04-20 by wes-b

sechternacht, 2020-04-22:
Hi wes-b,

Thank you for your help! I was under the impression I was working as Administrator and had permissions enabled, but they had been altered for that file.

As for the auto mode bug, I'm new to github and don't quite understand.   Do I need to execute the commands on the PR you linked in order to fix the issue, or will this be fixed in a future update?  Let me know and thank you again for your assistance.

wes-b, 2020-04-22:
Correct, the next release of the SDK will include the fix

PR #1179 (2020-04-17): K4aRecorder updated to not set gain to auto 
## Fixes #1165

### Description of the changes:
- Removed code to set auto gain - that is not supported
- Added debug info

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [x] I ran the [functional tests](https://g [... cut]