microsoft/Azure-Kinect-Sensor-SDK #1166: Transform engine create and initialize failed with error code: 204.

RachelZheng, 2020-04-08:
---
name: Transform engine create and initialize failed with error code: 204.
about: returned failure in transformation_create()
title: ''
labels: Bug, Triage Needed
assignees: ''

---

**Describe the bug**

I tried to use the k4a_capture_get_depth_image() and k4a_capture_get_color_image() to capture the video from playback. I installed the sensor library on Linux ubuntu 18.04 (also do the same at 16.04 but also fail) and ran my code. I can get k4a_capture_get_depth_image() from the mkv file I recorded from kinect, but I cannot get the color image from the same capture.

The code part is:

```
char *filename = "/home/ubuntu/data/test.mkv";
k4a_record_configuration_t record_config;
k4a_playback_t *handle = (k4a_playback_t *) malloc(sizeof(k4a_playback_t));
k4a_result_t result = K4A_RESULT_SUCCEEDED;
k4a_transformation_t transformation = NULL;
k4a_calibration_t calibration;
k4a_capture_t capture = NULL;
k4a_image_t depth_image = NULL;
k4a_image_t color_image = NULL;
k4a_stream_result_t stream_result;
k4a_image_t uncompressed_color_image = NULL;
result = k4a_playback_open(filename, handle);
int timestamp = 0;

if (result != K4A_RESULT_SUCCEEDED){
    printf("Failed to open file: %s\n", filename);
}

result = k4a_playback_seek_timestamp(*handle, timestamp * 1000, K4A_PLAYBACK_SEEK_BEGIN);
if (result != K4A_RESULT_SUCCEEDED)
{
    printf("Failed to seek timestamp %d\n", timestamp);
}
printf("Seeking to timestamp: %d/%d (ms)\n",
       timestamp,
       (int)(k4a_playback_get_recording_length_usec(*handle) / 1000));

// get the transfer handler
result = k4a_playback_get_calibration(*handle, &calibration);
if (result != K4A_RESULT_SUCCEEDED){
    printf("Failed to get calibration\n");
}

transformation = k4a_transformation_create(&calibration);

k4a_record_configuration_t config;

stream_result = k4a_playback_get_next_capture(*handle, &capture);
if (stream_result != K4A_STREAM_RESULT_SUCCEEDED || capture == NULL)
{
    printf("Failed to fetch frame\n");
}

result = k4a_playback_get_record_configuration(*handle, &config);
/* the configuration of the mkv files:
{color_format = K4A_IMAGE_FORMAT_COLOR_MJPG, 
  color_resolution = K4A_COLOR_RESOLUTION_3072P, 
  depth_mode = K4A_DEPTH_MODE_WFOV_UNBINNED, camera_fps = K4A_FRAMES_PER_SECOND_15, 
  color_track_enabled = true, depth_track_enabled = true, ir_track_enabled = true, 
  imu_track_enabled = false, depth_delay_off_color_usec = 0, 
  wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE, subordinate_delay_off_master_usec = 0, 
  start_timestamp_offset_usec = 120288}
 */

depth_image = k4a_capture_get_depth_image(capture); // success
color_image = k4a_capture_get_color_image(capture); // fail, color_image = 0x0;
```

**To Reproduce**

```
# install the sensor sdk
git clone git@github.com:microsoft/Azure-Kinect-Sensor-SDK.git
cd scripts/docker
chmod +x setup-ubuntu.sh
cd ../
mkdir build && cd build
cmake .. -GNinja
ninja
sudo ninja install

# add the libdepth library
# I download the deb file here:
wget https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4-dev/libk4a1.4-dev_1.4.0_amd64.deb  
# unzip, move the libdepthengine.so.2.0 to /usr/lib/x86_64-linux-gnu/

# install opencv
git clone https://github.com/opencv/opencv.git
cd opencv
mkdir build && cd build
cmake ../
make && make install
```

**Expected behavior**

I expect the color image to be captured from the mkv.

**Logs**

Seeking to timestamp: 0/10129 (ms)
[2020-04-08 19:48:57.826] [error] [t=29345] ../src/tewrapper/tewrapper.c (61): transform_engine_start_helper(). Transform engine create and initialize failed with error code: 204.
[2020-04-08 19:48:57.826] [error] [t=29345] ../src/tewrapper/tewrapper.c (68): teresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in transform_engine_start_helper()
[2020-04-08 19:48:57.826] [error] [t=29345] ../src/tewrapper/tewrapper.c (86): transform_engine_start_helper(tewrapper) returned failure in transform_engine_thread()
[2020-04-08 19:48:57.826] [error] [t=29344] ../src/tewrapper/tewrapper.c (313): tewrapper_create(). Transform Engine thread failed to start
[2020-04-08 19:48:57.826] [error] [t=29344] ../src/transformation/transformation.c (637): transformation_context->tewrapper != NULL returned failure in transformation_create()

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.4.0
 - Firmware version:  Not installed since it is for postprocessing of mkv file

**Additional context**
I checked the mkv file from the image viewer, and I could manually get the color images. I can also use ffmpeg to get the color image frames so I am sure the mkv file is okay. I also tried to copy the libdepthengine.so to Azure-Kinect-Sensor-SDK/build/bin/.

wes-b, 2020-04-13:
Error code 204 is K4A_DEPTH_ENGINE_RESULT_FATAL_ERROR_GPU_FROM_API

wes-b, 2020-04-13:
What is your OpenGL version? See this comment: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/973#issuecomment-571261360. Can you share the full output?

RachelZheng, 2020-04-13:
Hi, I am using AWS instance so when I type `glxinfo` it reminds me `Error: unable to open display`

wes-b, 2020-04-13:
Thank you, this is #810

closed 2020-07-11 by qm13

zoe2718, 2023-08-02:
Hi @RachelZheng ，I have the same problem. Could you tell me how you finally solved this problem?