microsoft/Azure-Kinect-Sensor-SDK #1168: Documentation / advice on how to process position values when camera is placed at an angle

immerse-ar, 2020-04-09:
When camera is tilted downward at an angle to do body tracking, positions of the joints are relative to the camera. This makes it difficult to interpret gestures and positions.

I was wondering if someone can point me in the correct direction on how to use gyroscope or IMU data from the kinect to transform the position values to emulate the joint positions as if the camera was not rotated.

Example use case: camera is positioned downward and when user raises their right wrist
Expected (desired) behavior: Wrist joint position goes from (0, 0, 0) -> (0,1,0)
Actual behavior: Wrist joint position goes from (0,0,0) -> (0, 0.75, 0.5)

I tried the formulas listed here to transform "y" and "z" positions for a rotated camera
https://www.cs.helsinki.fi/group/goa/mallinnus/3dtransf/3drot.html 

To achieve this, I placed the camera at about a 60 degree angle facing downward (effectively only rotated on the x-axis) and applied the following functions in the link above to the "y" and "z" positions.

JpEncausse, 2020-04-10:
The most simple way to detect gesture is to compare 2 elements from the camera reference.
For instance if you want to compare the raised right wrist, you should compare it with an other body joint (like above elbow).

If you compare it with 0,1,0 (so outside of the camera reference) you'll have to add the gyroscope data. I don't already play with it.

immerse-ar, 2020-04-10:
@JpEncausse we also want to detect location of the joints as well so we can emulate a person "walking around" in a digital environment, When the camera is at an angle though, the person walks in angles.

Sounds like the gyroscopic data is the way to go but I'll keep this open in case I can get some example code from the kinect team.

qm13, 2020-04-17:
Using the IMU to determine the camera rotation is the preferred approach assuming the camera is stationary. This is one of the reasons we included an IMU in the camera. The IMU is part of the extrinsic calibration.

You could also calibrate the cameras position using a marker board. Finally there is an RGBD SLAM project that can provide camera pose trajectory assuming the camera is moving https://github.com/ETH3D/badslam.

closed 2020-05-21 by qm13