microsoft/Azure-Kinect-Sensor-SDK #1169: What is the technical difference between Azure Kinect and Kinect V2's body tracking?

Jessica-Yeung, 2020-04-10:
After using the body tracking of Azure Kinect. We found most of the time, the Kinect V2 has better performance (speed, efficiency, and reliability) than it. So what is the technical difference between Azure Kinect and Kinect V2's body tracking except the Azure Kinect uses a DNN model to derive the skeleton, the Kinect v2 uses a ‘random forest’ model? In which situation we choose to use the Azure Kinect's body tracking instead of that of Kinect V2?

fractalfantasy, 2020-04-11:
From what I understand, the Azure Kinect's current Body Tracking SDK uses a machine learning model trained on a large data sets to extrapolate poses from the IR feed of the sensor. This approach has been quite successful for [Google](https://storage.googleapis.com/tfjs-models/demos/posenet/camera.html), [Intel](https://www.intelrealsense.com/skeleton-tracking/), [WrenchAI](https://www.youtube.com/watch?v=vTC0QKR_uM0), whereas Microsoft's attempt leaves a lot to be desired.

You can learn about the Kinect v2 Body Tracking random forest model [here](https://onlinelibrary.wiley.com/doi/full/10.4218/etrij.13.2013.0063).

In this moment the Kinect v2 Body Tracking is superior to it's successor in pretty much all respects but I'm rooting for the developer team to get the DNN body tracking model up to speed with the other companies - it has the potential to work even better than the others because it is trained on IR rather than RGB which is superior in dark environments. 

Azure Kinect was built for industrial applications, most of which rely on small form factor computers without discreet GPUs running passively all day.. so the developer team have the incentive to make body tracking work in these scenarios, but I think it'll take them a long time to get a fast, accurate and efficient DNN model up and running. Until then we're stuck with extremely latent and slow body tracking that only runs on Nvidia gaming rigs. 

Much of the developer community have  put their projects on hold because of this so I've started a [feature request](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2) to implement a Legacy Body Tracking option. This would give developers the choice to to use random forest Kinect v2-style body tracking over DNN if they want - the feature request has already amassed a lot of support so hopefully we'll have a solution sooner than later.

qm13, 2020-04-14:
@Jessica-Yeung the team debated porting the previous Kinect for Windows body tracking algorithm to Azure Kinect vs leveraging 10+ years of research into pose estimation since we developed the original algorithm. The team opted to develop a new model built around a DNN and plan to evolve this model to improves its performance and accuracy.

closed 2020-04-14 by qm13

rfilkov, 2020-04-16:
@fractalfantasy I think the first thing that needs to be requested is the Body Tracking SDK team to be replaced with another one, more skilled, experienced and responsive to customer requests.

dotslinker, 2020-04-21:
The current Body Tracking SDK is almost unusable on most of the devices where Kinect V2 was running smoothly. The result of this, especially in these Covid-19 times, is that we almost cannot use the new Kinect for Azure for all those applications home-based that are currently desperately needed to keep people active and in shape. Let me humbly suggest to the development team to consider opening a development branch of the previous SDK for Kinect v2 to be used with the new Kinect for Azure, of course paying the price of a less accurate tracking. Which for most of the interactive applications was already fairly good (we are ahppily using Kinect v2 + old sdk in a rehab center for children). 
Other solutions (i.e. based on Intel Realsense D425 or D415), much simpler and less effective from many points of view, are already in place, and if there will be no changes in the current Kinect for Azure SDK development plans regards the use of GPU-intesive DNN for the BT, I think many will switch to those ones .

fractalfantasy, 2020-04-21:
@dotslinker I agree with you on all fronts - this topic got closed but body tracking performance is now being tracked in @rfilkov’s thread #514 so feel free to copy paste your response there if you‘d like the sdk team to see it.

Also the idea you mentioned is on the feedback forum so please upvote if you’d like to see it happen:
https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2

dotslinker, 2020-04-22:
@fractalfantasy thanks a lot. I did it right now. It is really unbelievable that I may reach less than 20 fps for the BT SDK on a Dell i9 32 GB Ram just because I have a low rated Nvidia GTX 1650.
And the other thing I really don't understand is that usually the newer versions of the SDKs are set on the top of the previous ones, while the new K4W SDK one seems to be rather built from scratch and definitively less usable for many aspects (i.e. source examples just to name one) if compared to the Kv2...

Fulkeg, 2020-07-28:
Is it possible that the same virtual / ai modeling that the Azure uses for presentations could be achieved by the xbox one kinext?