microsoft/Azure-Kinect-Sensor-SDK #1174: Unity HDRP causes body tracking to stop working.

joel1st, 2020-04-14:
**Describe the bug**
Updating the Unity `sample_unity_bodytracking` project to use HDRP  (High Definition Render Pipeline) causes body tracking to stop working. Bodies are no longer picked up by the Kinect. `currentFrameData.NumOfBodies` always equals `0`. 

**To Reproduce**
After testing on multiple computers + multiple different versions of Unity, this appears to happen consistently on all tested versions (2019.3.19, 2018.4.18, 2020.1.0). 

1. Set up the `sample_unity_bodytracking` project following README.
2. Confirm project runs as expected and bodies are picked up.
3. Update scene to HDRP (eg. https://evermotion.org/tutorials/show/11342/upgrade-your-graphics-in-unity-with-hdrp-render-pipeline)
4. Rerun project and note lack of bodies being picked up. 

**Expected behavior**
I would expect that changing the rendering pipeline should not stop Azure Kinect body tracking from working. 

**Desktop (please complete the following information):**
 - Body Tracking SDK Version:  1.0.1

**Additional context**
There is an implementation that doesn't use the .NET body tracking wrapper here: that works with HDRP. The only issue with this HDRP implementation is that it is very laggy after a body is tracked.
https://github.com/satoshi-maemoto/Azure-Kinect-Unity-Body-Tracker/tree/branches/%2331_hdrp

qm13, 2020-04-25:
Are you using the official C# wrapper? Are you using any Azure Kinect Unity plug-in or have you implemented this yourself?

joel1st, 2020-04-25:
I am just using the sample provided by Microsoft. The only change is updating it to to HDRP
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking

anantley, 2020-04-27:
Hi @joel1st 
I just upgraded to HDRP using these instructions.  
https://blogs.unity3d.com/2018/09/24/the-high-definition-render-pipeline-getting-started-guide-for-artists/
I uninstalled Multiplayer HLAPI and re-installed it to version 1.0.4 
After these steps the Body Tracking worked.

Let me know if these steps work for you or if this is not what you trying to do

Angus

closed 2020-05-06 by qm13

joel1st, 2020-05-07:
Hey @anantley
Thanks for the reply. Unfortunately I don't have the Kinect with me any more to test it out. If I have another project that requires one, I'll give it a go.