microsoft/Azure-Kinect-Sensor-SDK #1181: sync ports missing on multiple devices

themancalledjakob, 2020-04-17:
**Steps to reproduce the error**
order kinect from https://www.microsoft.com/de-de/p/azure-kinect-dk/8pp5vxmd9nhq?activetab=pivot%3aoverviewtab

**Expected results**
receive kinect DK according to hardware specifications

**Actual results**
all ordered Kinects are missing sync-in and sync-out ports (see image)
[image]

**SDK version**
irrelevant

**Firmware version**
irrelevant

**Log Results from the repro (if possible)**
irrelevant

I was hoping maybe someone can explain to me what is happening here? I have now seven of these cameras, and none of them has a sync port.
I can't find anything about missing sync ports online.
They were ordered last week, and should be current versions. 
Are they able to sync now via USB, or do I need to send all of them back now?

This is very confusing and freaks me out a bit.
I realize, that this is not a code issue, and this might not be the right place to post this issue. But I tried to reach support for a while now, and either do not get connected or called back, and I don't want to wait until I can't return them anymore..

Brekel, 2020-04-17:
Remove both screws in the back with the included tool, slide the cover off and you'll find the sync ports underneath.

themancalledjakob, 2020-04-17:
OMG!
Now I feel stupid but also very happy. Thank you!

closed 2020-04-17 by themancalledjakob