microsoft/Azure-Kinect-Sensor-SDK #1183: I want to work my own Avatar through BodyTraking SDK in Azure Kinect

SMLimtsb, 2020-04-20:
I use Azure Kinect DK to move my own 3D object Avatar through BodyTracking SDK in Unity.

I tried, but each Joint is wrong place like a trash.

(I refer to joint coordinates this : https://docs.microsoft.com/en-us/azure/kinect-dk/body-joints)

(and It is very similar with my situation : Kinect v2 avatar controller with unity)

To Solve this, Can I make my own 3D object avatar matching to Azure Kinect's Skeleton data for following user?

Let me know how can I make 3D obejct with Maya if I can.

or Should I use Avateering? if I want to use like that.

Is there any sample about Avateering of azure kinect with unity?

anantley, 2020-05-07:
Hi Have you tried using the sample unity application in this repository?
https://github.com/microsoft/Azure-Kinect-Samples.git

Here is a new script that may solve your problem: 
https://gist.github.com/anantley/fd02b3facd372317abda668e58d7cae6

To use this script add it to the Assets of the sample_unity_bodytracking project in Azure Kinect Samples repository.
https://github.com/microsoft/Azure-Kinect-Samples.git
1)Then import a character into your scene, make sure it is humanoid.
2)Add this script to the character.
3)Make the kinect device your kinect4AzureTracker prefab
4)Make the root the pointbody/pelvis joint
5) Set the offset if you want to

If you could try this and let me know it would be great to hear your feedback.
Thanks,
Angus

SMLimtsb, 2020-05-08:
Confirm Page

.table1
{
	color: #000000;
	text-decoration: none;
	margin:0;
	padding-left:5px;
	padding-top:4px;
	padding-bottom:4px;
	padding-right:5px;
	line-height:17pt;
	font-family: "GulimChe";
	font-size: 12px;
	font-weight: normal;
	border-top:0px;
	border-left:0px;
	border-right:0px;
	border-bottom:0px;

}

   
    
    
    
    
    
    
  
   
    
    
    
	
    
    
    
  
   
    
    
    
    
    
    
  
   
    
    
    
    
    
  
   
    
    
    편지 감사합니다.
      저는 sm.lim@tsb.co.kr 이메일주소 사용자 입니다
      저에게 편지를 보내시려면 처음 한 차례의 인증이 완료되어, 저의 주소록에 귀하가 등록이 되어야만, 저에게 정상적으로 메일을 전송할 수 있습니다. 
본 인증 이후 저의 주소록에 등록되면,그 후 발송되는 메일은 인증과정이 생략됩니다
본 메일은 스팸메일차단을 위한 인증메일입니다. 
따라서 본 메일의 제목, 송신자 이메일주소, 내용 등을 확인한 후, 귀하가 보낸 메일에 대해 발송된 것이 아니면 절대 인증을 하지마시기 바랍니다.
	
    
  
   
    
     
    
     
    
  
   
    
    
    
    
    
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
    
  
   
    
    
        경고: 60일 이내에 인증번호를 전송하지 않을 경우, 보관된 메일이 삭제되오니, 본 메일 확인 후, 꼭 인증번호를 입력해 주시기 바랍니다. 
	
    
  
   
    
    
    
    
    
    
  
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
      

  
   
    
    
    
      This is sm.lim@tsb.co.kr. 
      Please complete the verification so that I can receive your email. 
	“After finishing this verification, 
	you can be in my address book forever and won't need to do it again.” 
	This is the confirmation e-mail for blocking spam e-mail.
	If this isn't e-mail for you after checking the e-mail title, sender's e-mail address and body, don't finish the confirmation process. 
	
    
  
   
    
     
    
     
    
  
   
    
    
    
    
    
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
    
  
   
    
    
	  Warning: you must confirm to deliver the mail, or it will be droped in 60 days!

SMLimtsb, 2020-05-10:
thank you. I tried that sample immediately.
sorry, but I have a problem to try the sample project like this.

> AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
> Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, Microsoft.Azure.Kinect.BodyTracking.TrackerConfiguration configuration) (at D:/a/1/s/src/csharp/Microsoft.Azure.Kinect.BodyTracking/Tracker.cs:32)
> SkeletalTrackingProvider.RunBackgroundThreadAsync (System.Int32 id) (at Assets/Scripts/SkeletalTrackingProvider.cs:35)   ..........

I guess this code is not working.

> Tracker tracker = Tracker.Create(deviceCalibration, new TrackerConfiguration() { ProcessingMode = TrackerProcessingMode.Gpu, SensorOrientation = SensorOrientation.Default });

what can I do to solve this problem?

anantley, 2020-05-13:
Usually this is a sign the dependencies are not in the right places.

1) clone this repo https://github.com/microsoft/Azure-Kinect-Samples.git
2) open the sample_unity_bodytracking project that is in this repo using Unity version 2019.1 or later
3) open the associated visual studio project by clicking on a script in the assets in the editor
4) open the tools -> NuGet package manager console
5) type this: Install-Package Microsoft.Azure.Kinect.BodyTracking -Version 1.0.1
6) then run the MoveLibraryFiles.bat file in the project root directory
7) then you should be able to run your application
You can check your install by reading the README.md in the project root if you want to be sure that everything is there.

Angus

SMLimtsb, 2020-05-13:
Confirm Page

.table1
{
	color: #000000;
	text-decoration: none;
	margin:0;
	padding-left:5px;
	padding-top:4px;
	padding-bottom:4px;
	padding-right:5px;
	line-height:17pt;
	font-family: "GulimChe";
	font-size: 12px;
	font-weight: normal;
	border-top:0px;
	border-left:0px;
	border-right:0px;
	border-bottom:0px;

}

   
    
    
    
    
    
    
  
   
    
    
    
	
    
    
    
  
   
    
    
    
    
    
    
  
   
    
    
    
    
    
  
   
    
    
    편지 감사합니다.
      저는 SM.Lim@tsb.co.kr 이메일주소 사용자 입니다
      저에게 편지를 보내시려면 처음 한 차례의 인증이 완료되어, 저의 주소록에 귀하가 등록이 되어야만, 저에게 정상적으로 메일을 전송할 수 있습니다. 
본 인증 이후 저의 주소록에 등록되면,그 후 발송되는 메일은 인증과정이 생략됩니다
본 메일은 스팸메일차단을 위한 인증메일입니다. 
따라서 본 메일의 제목, 송신자 이메일주소, 내용 등을 확인한 후, 귀하가 보낸 메일에 대해 발송된 것이 아니면 절대 인증을 하지마시기 바랍니다.
	
    
  
   
    
     
    
     
    
  
   
    
    
    
    
    
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
    
  
   
    
    
        경고: 60일 이내에 인증번호를 전송하지 않을 경우, 보관된 메일이 삭제되오니, 본 메일 확인 후, 꼭 인증번호를 입력해 주시기 바랍니다. 
	
    
  
   
    
    
    
    
    
    
  
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
      

  
   
    
    
    
      This is SM.Lim@tsb.co.kr. 
      Please complete the verification so that I can receive your email. 
	“After finishing this verification, 
	you can be in my address book forever and won't need to do it again.” 
	This is the confirmation e-mail for blocking spam e-mail.
	If this isn't e-mail for you after checking the e-mail title, sender's e-mail address and body, don't finish the confirmation process. 
	
    
  
   
    
     
    
     
    
  
   
    
    
    
    
    
  
   
    
    
    
     
    
    
  
   
    
    
    
    
    
    
  
   
    
    
	  Warning: you must confirm to deliver the mail, or it will be droped in 60 days!

SMLimtsb, 2020-05-18:
I still have a same problem after following that all lines.
I can see "already exist package of Project" in line 5.
and I saw moving LibraryFiles when I follow line 6.
So it still a same error message in unity console like 

> AzureKinectBodyTrackingCreateException: result = K4A_RESULT_FAILED
> Microsoft.Azure.Kinect.BodyTracking.Tracker.Create (Microsoft.Azure.Kinect.Sensor.Calibration calibration, [...]

I used Unity version 2019.1.11f.

what am i do?

anantley, 2020-05-19:
Ok I just tried with 2019.1.2, which is an earlier version and it worked.  So it's not the Unity version.
Could you try this command on the package manager console command line?

UnInstall-Package Microsoft.Azure.Kinect.BodyTracking -Version 1.0.1
Then delete the Assets/Plugins folder

Followed by:
Install-Package Microsoft.Azure.Kinect.BodyTracking -Version 1.0.1
and then run the MoveLibraryFiles.bat file

Let me know if that works.

Thanks,
Angus

SMLimtsb, 2020-05-20:
I was difficult to work without problem. but I follow you from completely first step.

so It works very well with PuppetAvatar. thank you!

closed 2020-05-20 by SMLimtsb

nmeri12, 2020-11-03:
Hi @anantley,

I already followed your instructions but the 3D Avatar Character didn't work, so if you could help me please!
Also, when I download the project it shows that Female_Adult_11 (Missing Prefab), also jasper (Missing Prefab).
I delete it and try to import it gain or with different character but also didn't work
I followed the following steps:
1)Then import a character into your scene, make sure it is humanoid. (I am sure it is, I did also the Reg and the Joints)
2)Add this script to the character.
3)Make the kinect device your kinect4AzureTracker prefab
4)Make the root the pointbody/pelvis joint (do u mean Root Position or Character Root, although I tried them both but no working)
5) Set the offset if you want to

Best Regards,
M.Nmeri

anantley, 2020-11-03:
Hi @nmeri12,  can you send me a screen capture of what is happening?  Is there no tracking at all? Or are you having incorrect tracking?

Also a screen capture of the hierarchy of the character you are using would help.

If you need a character to puppet you can get some by cloning this repository: 
https://github.com/microsoft/Microsoft-Rocketbox.git

Best,
Angus

nmeri12, 2020-11-04:
Hi @anantley 
thank you for your response.

[image]

[image]

[image]

The avatar character is from Mixamo. The problem is that no tracking at all (see photo 2)
Note: the script 

best,
M.Nmeri

nmeri12, 2020-12-12:
Hi @anantley,

sorry for the delay answer but I tried all possible ways and I have the avatar but it's nor working correctly. I still have a problem with the head and hands by playing the scene. They do not move correctly. I tried to use different avatars but I think the problem is in the Tracker (Kinect4AzureTracker) or in the Avatar script.

I will be glad if u could help me and give me some advices.

[image]

[image]

or if u can tell me how I can remove the white balls(Head) in the Tracker

[image]

is it correct like that 👍
[image]

One thing more which script is correct the one here (https://github.com/microsoft/Azure-Kinect-Samples/blob/master/body-tracking-samples/sample_unity_bodytracking/Assets/Scripts/PuppetAvatar.cs)
or the one here (https://gist.github.com/anantley/fd02b3facd372317abda668e58d7cae6)
cause there is a different between both and I tried them both the first one doesn't work.
Best regards,
m.n

pmelio01, 2021-02-08:
Hello @anantley ,

I have a similar problem. I imported my character to the scene, I also added the script, but it's not working.

any ideas?

anantley, 2021-02-08:
Did you make the character a humanoid in Unity?

pmelio01, 2021-02-09:
By default my character was set as a 'Generic'. 
Setting it to 'Homanoid' it got working. 
Thank you