microsoft/Azure-Kinect-Sensor-SDK #1184: Support x86 version of Azure Kinect Body Tracking SDK

jerogor, 2020-04-20:
**Is your feature request related to a problem? Please describe.**

The Azure Kinect Body Tracking SDK binaries are currently only published as 64-bit dlls. To enable 32 bit applications to use this SDK they need access to 32 bit versions of the binaries. 

**Describe the solution you'd like**

If possible, include 32 bit versions of the dlls and k4abt.lib for the Azure Kinect Body Tracking SDK as it was done for the Azure Kinect Sensor SDK : https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/386.

**Describe alternatives you've considered**

**Additional context**

I am currently developing an application which must be 32 bit to use a device (ultrasound probe) with Kinect Azure and especially body tracking. This application will run in 64 bit windows environment and works fine with the 32 bit version of Azure Kinect Sensor SDK but I would need support of 32 bit version of the Azure Kinect Body Tracking SDK to perform body tracking within this 32 bit application.

qm13, 2020-04-27:
Unfortunately we are not in a position to provide a 32bit binary. We are reliant on two third party components - CUDA and ONNX Runtime - that are not available as 32bit binaries. There is a thread on stackoverflow - https://stackoverflow.com/questions/58828349/x86-32-bit-support-for-cuda - that covers the CUDA limitation.

jerogor, 2020-04-29:
Thank you for your answer. By the way for those interested, I overcomed this problem by compiling my app in 64 bit to use Kinect Azure Body Tracking SDK and within this app I launch a 32 bit executable which gives me the ultrasound images I needed thanks to a QProcess from the Qt library. Pretty ugly... but it works!

closed 2020-04-29 by jerogor