microsoft/Azure-Kinect-Sensor-SDK #1185: Multi-Device Sync Issue with Depth Frames

DavidRobb, 2020-04-20:
I have two Devices connected via a sync cable as Master and Slave and have problems relating the frames produced.  I believe the frames are synced in that the data is captured at approximately the same time. However, the problem arises is that the actual frames read from the devices is not always consistent presumably due to the internal buffering of frames. Sometimes( not always but most times) the frames read from the slave device are two frames behind what I read from the Master device. Since the devices run independent clocks, how am I meant to correlate the two streams? ( I can tell the frames are out as my two point clouds are aligned, however on waving a hand in front of the cameras, the moving portion is not aligned by two frame steps)

I believe I have followed all the correct instructions for syncing multiple devices:
I'm using the latest firmware and SDK 1.4.0 (also seen on 1.3.0)
I start the slave device first and then the master.
I read the slave device first and then the master.
I'm using manual exposure (minimal 500us time as I'm only interested in the depth frames)
I have an offset of 160us to avoid interference.
Bug reproducible in Windows and Linux.

Here is my camera initialisation code:
```C++
bool DeviceKinect::prepare() {
    _log->info("Prepare device {} ", m_camId);

    // Manual exposure time recommended for synced depth mode
    int32_t exposureValueUs = 500;
    int32_t fiftyHz = 1;
    k4a_device_set_color_control(m_device, K4A_COLOR_CONTROL_POWERLINE_FREQUENCY, K4A_COLOR_CONTROL_MODE_MANUAL, fiftyHz);
    k4a_device_set_color_control(m_device, K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, K4A_COLOR_CONTROL_MODE_MANUAL, exposureValueUs);

    m_config.depth_mode = K4A_DEPTH_MODE_WFOV_2X2BINNED;
    m_config.camera_fps = K4A_FRAMES_PER_SECOND_30;
    m_config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
    m_config.color_resolution = K4A_COLOR_RESOLUTION_720P;
    m_config.synchronized_images_only = false;
    m_config.depth_delay_off_color_usec = 00;

    if (K4A_RESULT_SUCCEEDED == k4a_device_get_sync_jack(m_device, &syncInConnected, &syncOutConnected)) {
        if (syncInConnected && !syncOutConnected) {
            _log->info("Sync cable is slave for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
            //            m_config.color_resolution = K4A_COLOR_RESOLUTION_OFF; // Cannot be off as cannot then re-start camera!
            m_config.subordinate_delay_off_master_usec = 160; // Recommended min of 160uS to avoid multiple cam interference
        } else if (!syncInConnected && syncOutConnected) {
            _log->info("Sync cable is master for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
        } else {
            _log->error("Sync cables not correctly connected for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
        }
    } else {
        _log->error("Failed to determine Sync cable state for cam {} ", m_camId);
        m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
    }

    if (K4A_RESULT_SUCCEEDED != k4a_device_get_calibration(m_device, m_config.depth_mode, m_config.color_resolution, &m_calibration)) {
        _log->warn("Failed to get calibration");
        return false;
    }

    k4a_color_control_mode_t mode;
    int32_t exposeTime;
    int32_t powerFreq;

    if (K4A_RESULT_SUCCEEDED != k4a_device_get_color_control(m_device, K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, &mode, &exposeTime)) {
        _log->warn("Failed to get EXPOSURE TIME value");
        return false;
    }
    _log->info("Exposure time is {}us for cam {}", exposeTime, m_camId);

    if (K4A_RESULT_SUCCEEDED != k4a_device_get_color_control(m_device, K4A_COLOR_CONTROL_POWERLINE_FREQUENCY, &mode, &powerFreq)) {
        _log->warn("Failed to get Power Freq value");
        return false;
    }
    _log->info("Power line freq is {}Hz for cam {}", powerFreq == 1 ? "50" : "60", m_camId);

    k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM, m_calibration.depth_camera_calibration.resolution_width,
                     m_calibration.depth_camera_calibration.resolution_height,
                     m_calibration.depth_camera_calibration.resolution_width * (int)sizeof(k4a_float2_t), &m_xy_table);

    create_xy_table(&m_calibration, m_xy_table);

    return true;
}

![image](https://user-images.githubusercontent.com/11753801/79793930-4dd6de80-8349-11ea-95a3-3b7c432d779b.png)

wes-b, 2020-04-20:
The main thing to do is start the subordinate first, which it sounds like you are doing that.  The device timestamp is the one that is synchronized, not the system timestamp. Take a look at get_synchronized_captures in https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/a144b6fbe382c6357d74158414e703388d2e2608/examples/green_screen/MultiDeviceCapturer.h#L95 for how we do it in the green screen example.

We also have gtest you can run if you have 2 camera's connected to confirm everything is synchronizing as expected. Run multidevice_ft.exe after building the SDK, you need 2 sync cables and 2 cameras for it to work. Hook them up such that either camera can be master or subordinate.

DavidRobb, 2020-04-21:
Many thanks for the link to the MultiDeviceCapturer sample. From the API documentation I assumed that the device timestamp was simply an independent rolling us timer present on each camera and used to stamp the images. I had no idea that when the devices are connected via sync cables that I can compare the device timestamp from one camera with that from another camera. I can see an easy way forward now. Thanks. 
Maybe the documentation just needs improving to indicate this?

wes-b, 2020-04-21:
We are all ears when it comes to making documentation better. Can you tell me what documentation you consulted that we could update?

We have this topic that should have been easy to find https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync.

I am closing the issue, but we are still listening

closed 2020-04-21 by wes-b