microsoft/Azure-Kinect-Sensor-SDK #1187: Error with DecodeMJPEGtoBGRA32: MJPEG decode failed: -1

omartin2010, 2020-04-20:
**Describe the bug**
When I use the SDK, sometimes (pretty often since this morning), I get this error :  MJPEG decode failed: -1

**To Reproduce**
I don't unfortunately have a clear repro. My problem occurs now very often, so in my project I can repro most of the time since this morning.

**Expected behavior**
I expect this to work like it did before.

**Logs**
I've enabled the `info` logging level.
```
[2020-04-20 16:29:59.465] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (343): find_libusb_device(). Container ID found: {107b61c6-b351-449a-63bd-03c8c35383ec}
[2020-04-20 16:29:59.466] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (224): populate_serialnumber(). Serial Number found 000211394812
[2020-04-20 16:29:59.576] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (343): find_libusb_device(). Container ID found: {107b61c6-b351-449a-63bd-03c8c35383ec}
[2020-04-20 16:29:59.577] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (224): populate_serialnumber(). Serial Number found 000211394812
[2020-04-20 16:29:59.605] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (225): ******************** Device Info ********************
[2020-04-20 16:29:59.605] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (226): K4A SDK version:     1.4.0
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (232): Serial Number:       000211394812
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (236): RGB Sensor Version:  1.6.108
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (239): Depth Sensor Version:1.6.79
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (242): Mic Array Version:   1.6.14
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (245): Sensor Config:       6109.7
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (248): Build type:          Release
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (253): Signature type:      MSFT
[2020-04-20 16:29:59.630] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (259): ****************************************************
[2020-04-20 16:30:00.631] [info] [t=43] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dynlib/dynlib_linux.c (125): dynlib_find_symbol(). Depth Engine loaded /usr/lib/aarch64-linux-gnu/libk4a1.4/libdepthengine.so.2.0
[2020-04-20 16:30:00.631] [info] [t=43] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/deloader/deloader.cpp (51): verify_plugin(). Loaded Depth Engine version: 2.1.3
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (868): k4a_device_start_cameras(). Starting camera's with the following config.
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (869): k4a_device_start_cameras().     color_format:3
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (870): k4a_device_start_cameras().     color_resolution:1
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (871): k4a_device_start_cameras().     depth_mode:4
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (872): k4a_device_start_cameras().     camera_fps:1
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (873): k4a_device_start_cameras().     synchronized_images_only:1
[2020-04-20 16:30:01.713] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (874): k4a_device_start_cameras().     depth_delay_off_color_usec:0
[2020-04-20 16:30:01.714] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (875): k4a_device_start_cameras().     wired_sync_mode:0
[2020-04-20 16:30:01.714] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (876): k4a_device_start_cameras().     subordinate_delay_off_master_usec:0
[2020-04-20 16:30:01.714] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (877): k4a_device_start_cameras().     disable_streaming_indicator:0
[2020-04-20 16:30:04.258] [info] [t=1] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (916): k4a_device_start_cameras(). k4a_device_start_cameras started
[2020-04-20 16:30:04.539] [info] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (277): capturesync_add_capture(). Dropped 2 depth captures waiting for time stamps to stabilize
[2020-04-20 16:30:04.571] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:    333611 type:Depth
[2020-04-20 16:30:05.002] [warning] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[2020-04-20 16:30:06.241] [info] [t=52] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (170): queue_pop(). Queue "Queue_capture" dropped oldest 1 captures from queue.
[2020-04-20 16:30:06.381] [info] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   2066877 type:Color
[2020-04-20 16:30:06.431] [info] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   2200222 type:Color
[2020-04-20 16:30:06.478] [info] [t=52] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (170): queue_pop(). Queue "Queue_capture" dropped oldest 2 captures from queue.
[2020-04-20 16:30:07.179] [info] [t=52] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (170): queue_pop(). Queue "Queue_capture" dropped oldest 1 captures from queue.
[2020-04-20 16:30:07.242] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   2933533 type:Color
[2020-04-20 16:30:07.267] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   3000255 type:Depth
[2020-04-20 16:30:07.320] [info] [t=52] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (170): queue_pop(). Queue "Queue_capture" dropped oldest 1 captures from queue.
[2020-04-20 16:30:07.910] [info] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   3666877 type:Color
[2020-04-20 16:30:08.039] [warning] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[2020-04-20 16:30:08.107] [info] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   3866866 type:Color
[2020-04-20 16:30:09.098] [warning] [t=45] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (278): depth_engine_thread(). Depth image processing is too slow at 1000ms (this may be transient).
[2020-04-20 16:30:09.190] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   4400200 type:Color
[2020-04-20 16:30:09.388] [info] [t=52] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (170): queue_pop(). Queue "Queue_capture" dropped oldest 11 captures from queue.
[2020-04-20 16:30:09.619] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   4466877 type:Color
[2020-04-20 16:30:09.631] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   4533611 type:Depth
[2020-04-20 16:30:09.631] [info] [t=50] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (86): drop_sample(). capturesync_drop, Dropping sample TS:   4866944 type:Depth
[... cut ...]
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**
I'm using this with the `develop` branch of the k4a python package found [here](https://github.com/etiennedub/pyk4a). Not sure this has a link to it as it worked for some time and then stopped working without the wrapper changing, nor the underlying SDK changing. I've updated the firmware and SDK from internal alpha to the versions on the package repo (this is AMD64) and both docker images I have do repro the issue.

omartin2010, 2020-04-20:
Any chance this can be related to memory exhaustion? I'm noticing some warnings outside of this specific docker container that my code may be not releasing memory properly... Just a thought.

wes-b, 2020-04-23:
This means the format of the image being decoded is corrupt. It either happens at the firmware level being read off the camera sensor or via memory corruption in the application. USB CRC's the data, so it is highly unlikely the issue happen there.

Dup of #926

omartin2010, 2020-04-23:
I’ll try to narrow get a repro. On the jetson, gpu memory is shared with main memory and I suspect this might be related... I’m retraining a smaller model to deploy on the jetson so that more ram is available to this container. Even if you limit ram available to the container, as I see on this platform, TensorFlow seems to see the full GPU memory available to it... without respect to the limit set in docker.

wes-b, 2020-04-24:
This issue was also recently raised here: https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/125

I have been able to repro this on a machine with 3 cameras, K4AViewer, 2160P, BGRA, 1.6.110079014 firmware. Frame rate was reported to be about 4FPS while running at 30.

Need to investigate if libjpeg-turbo has any fixes to this code.

Issue happens as PC becomes more CPU bound.

100% work around is to avoid BGRA32 format. 
Using BGRA32 with lower resolution reduces the chance of hitting this.

omartin2010, 2020-04-24:
I've noticed that higher CPU usage tends to repro more often. And higher memory. I've limited GPU memory available to tensorflow on the other container so that the main memory (on Jetson TX2 at least, main memory is shared with GPU memory) is not starved and it *seems* to improve. I will investigate using other formats - need to modify the underlying python binding. I'm using relatively 720P, much lower than 2160p and a single camera. But the Jetson is not super powerful. If required - I'm internal - I can provide some docker containers to repro (two, one is the app itself and the other is the model serving part - which consumes GPU/shared memory).

wes-b, 2020-04-28:
@carcormir from https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/125

>Hi @wes-b, sorry to bother you again but we are really trying to solve this issue since soon it will interfere with our workflow.
>
>As mentioned in one of my comments, the error comes from line 1288 in uvc_camerareader.cpp. In that code, the status of the decompression is only valid if it is 0. Otherwise it will return an error with no further checking.
In this issue Cannot open MJPEG image coming from Camera
#157 they mention that tjDecompress2 may return -1 if there is an error or a warning. They suggest to use a function tjGetErrorCode() to check if the error was a result of a warning or a fatal error.
>
> I would suggest you to investigate if it is possible to implement this in the SDK since the error that I am getting could potentially be a warning from tjDecompress2. From the external libraries that are used in the SDK, I can see that tjGetErrorCode() is available.

There are two things we can do to address this. The first is as you suggest, we can decode the error to better understand what is happening, we should definitely not be failing if a warning is generated. 

Second, I was able to repro the issue on Ubuntu AMD64, which happens when I start or stop a camera when there are 2 already running. I can repro this with any of the 4 color formats. So this leads me to believe that the host PC is falling behind in reading frames from the camera. When the camera runs out of resources it is forced to drop the frame actively being copied to the host. The host doesn't know the exact image size for MJPEG, so the SDK can't know the image is incomplete until it is decoded with tjDecompress2. I believe we just need to drop the frame in this situation and carry on instead of erroring out.

I can't say when a fix for this will be ready, but we are happy to have help on this if someone wants to submit a PR.

RoseFlunder, 2020-04-28:
> I believe we just need to drop the frame in this situation and carry on instead of erroring out.

Maybe print some kind of warning or something? I think the user should know that the host PC can't keep up with reading the frames if that would be the case? If so, how could this be solved without using addtional PCs? Is this a bottleneck for a single usb host controller setup and can be solved by adding additional usb  controllers via PCI Express cards or is there another solution?

carcormir, 2020-04-28:
Hi @RoseFlunder,

I tried to use an additional USB card with 4 extra independent controllers [(this one)](https://www.startech.com/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V) and I still get the same issue.
This are the specs of my PC and I would assume it should be powerful enough to handle everything, but maybe I am wrong.
PC specs:
Processor: AMD® Ryzen threadripper 2970wx 24-core processor × 48
2 GPUs: GeForce RTX 2080 Ti/PCIe/SSE2
Motherboard: ASRock x399 Taichi
Ram: 64Gb

omartin2010, 2020-04-28:
I'm seeing the issue on a much less powered, NVIDIA Jetson TX2, and that's with a single unit. I wouldn't mind skipping frames for my particular use case, and seeing the warning being raised is ok.

closed 2020-04-30 by wes-b

carcormir, 2020-05-01:
Hi @wes-b ,
I have tried the version with the fix #1194 and 3 cameras connected.
Everything seems to be working as expected after the fix. Thanks for being so fast!

After running the three instances for a few minutes I get a different error (see log file below)

This is the error
`[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS: 326682755 type:Depth`

It seems like (from the log file) that my GPU is not working hard enough. However my hardware consists of 2 GPUs GeForce RTX 2080 Ti/PCIe/SSE2, which I assume should do the job of keeping the processing under 33ms as you mentioned in #1112 

Any clue of why this could be happening? If you think it makes sense, you could open a new issue from this comment since it is not related to #1187 any more. 

[test_1_camera_k4a.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4564465/test_1_camera_k4a.log)

wes-b, 2020-05-01:
Great new on the fix, thank you for the confirmation! Can you open a new bug on this:

> After running the three instances for a few minutes I get a different error (see log file below)
>
> This is the error
> `[ error ] : replace_sample(). capturesync_drop, releasing capture early due to [...]

knat, 2020-05-01:
It's easy to convert YUY2 image to RGB image, just a few lines of code. See https://docs.microsoft.com/en-us/windows/win32/medfound/recommended-8-bit-yuv-formats-for-video-rendering#converting-8-bit-yuv-to-rgb888

PR #1194 (2020-04-30): Dropping failed conversions from MJPEG
## Fixes #1187 

### Description of the changes:
- Drops image that fails to be decoded, which happens on PC's that are heavily loaded and can't always read image from sensor fast enough

### I tested changes on: 
- [ ] Windows
- [x] Linux

Manual Testing will be done on Ubuntu with 3 camera's starting and stopping at random intervals. Expectation is that we will see warnings of dropped frames but no errors. 

NOTE: K4AViewer converts YUY and NV12 to bgra format for displaying purposes which are not being addressed. So those formats in that tool will stil [... cut]