microsoft/Azure-Kinect-Sensor-SDK #1188: "IR to Color" transformation missing

fractalfantasy, 2020-04-21:
Tried to get IR to color space conversion using k4a_transformation_depth_image_to_color_camera_custom, but it doesn't seem to work.

an "IR to Depth" transformation should be very easy to add - as it's pretty much exactly the same as the Depth to Color transformation.

rabbitdaxi, 2020-04-24:
@fractalfantasy the k4a_transformation_depth_image_to_color_camera_custom should be able to fulfill the purpose here. please provide details of how you are using this function and what issue/error you were hitting. We haven't receive any report of people having issue using this function yet.

mikkamikka, 2020-04-27:
@rabbitdaxi We trying to add [k4a_transformation_depth_image_to_color_camera_custom](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gac00dd00e7612a86382e3d0a130f276bb.html#gac00dd00e7612a86382e3d0a130f276bb)  IR to Color transformation to our DLL to be used in Electron app.
Here is the code snippet. I omit all the variables initialisation and ir, depth, color frames 'get' functions, as there's no problems with them.

`int ir_to_color_image_created = k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM16, jsFrame.colorImageFrame.width, jsFrame.colorImageFrame.height, jsFrame.colorImageFrame.width * 2, &ir_to_color_image);

	int depth_to_color_image_created = k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16, jsFrame.colorImageFrame.width, jsFrame.colorImageFrame.height, jsFrame.colorImageFrame.width * 2, &depth_to_color_image);

	if ( (ir_to_color_image_created == K4A_RESULT_SUCCEEDED) && (depth_to_color_image_created == K4A_RESULT_SUCCEEDED) )
	{

		jsFrame.irToColorImageFrame.height = 101; //debug
							
		if (k4a_transformation_depth_image_to_color_camera_custom(transformer, depth_image, ir_image, depth_to_color_image, ir_to_color_image,
			K4A_TRANSFORMATION_INTERPOLATION_TYPE_NEAREST , 0) == K4A_RESULT_SUCCEEDED)
		{
			jsFrame.irToColorImageFrame.width = 102; //debug

			jsFrame.irToColorImageFrame.width = k4a_image_get_width_pixels(ir_to_color_image);
			jsFrame.irToColorImageFrame.height = k4a_image_get_height_pixels(ir_to_color_image);

			jsFrame.irToColorImageFrame.image_length = k4a_image_get_size(ir_to_color_image);
			jsFrame.irToColorImageFrame.stride_bytes = k4a_image_get_stride_bytes(ir_to_color_image);
			jsFrame.irToColorImageFrame.image_data = new uint8_t[jsFrame.irToColorImageFrame.image_length];

		}
		else {
			jsFrame.irToColorImageFrame.width = 1;
		}
	}`

`k4a_transformation_depth_image_to_color_camera_custom` function never succeeded. Output `ir_to_color_image` data array is empty.
Custom image format is set to `K4A_IMAGE_FORMAT_CUSTOM16`

Similar `k4a_transformation_depth_image_to_color_camera(transformer, depth_image, depth_to_color_image)`  function works just fine in the same context.

What are we missing?

Sensor config:
depth_mode: KinectAzure.K4A_DEPTH_MODE_WFOV_2X2BINNED,
color_format: KinectAzure.K4A_IMAGE_FORMAT_COLOR_BGRA32,
color_resolution: KinectAzure.K4A_COLOR_RESOLUTION_720P,
camera_fps: KinectAzure.K4A_FRAMES_PER_SECOND_30

fractalfantasy, 2020-04-29:
@rabbitdaxi do you have an idea as to what could be wrong here? ^

rabbitdaxi, 2020-04-30:
@fractalfantasy sorry for the delay response (I am working on other projects now, so not looking github that frequent). 

I see the issue. The k4a_transformation_depth_image_to_color_camera API requires both input and output custom image has the same custom format, e.g. in your case, K4A_IMAGE_FORMAT_CUSTOM16. 

In your sample code, you passed the IR image directly into this API, the code internally has a check the format to matching got failed. I know it looks like a extra step, but based on how the SDK [expectation](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gac00dd00e7612a86382e3d0a130f276bb.html#gac00dd00e7612a86382e3d0a130f276bb), you can create a custom_ir_image as the input custom image with the matching format of the output custom image, see below code snippet:

`
k4a_image_t custom_ir_image = NULL;
        int ir_image_width_pixels = k4a_image_get_width_pixels(ir_image);
        int ir_image_height_pixels = k4a_image_get_height_pixels(ir_image);
        int ir_image_stride_bytes = k4a_image_get_stride_bytes(ir_image);
        uint8_t *ir_image_buffer = k4a_image_get_buffer(ir_image);
        if (K4A_RESULT_SUCCEEDED != k4a_image_create_from_buffer(K4A_IMAGE_FORMAT_CUSTOM16,
                                                                 ir_image_width_pixels,
                                                                 ir_image_height_pixels,
                                                                 ir_image_width_pixels * (int)sizeof(uint16_t),
                                                                 ir_image_buffer,
                                                                 ir_image_height_pixels * ir_image_stride_bytes,
                                                                 [](void *_buffer, void *context) {
                                                                     delete[](uint8_t *) _buffer;
                                                                     (void)context;
                                                                 },
                                                                 NULL,
                                                                 &custom_ir_image))
        {
            printf("Failed to create custom ir image\n");
            return false;
        }
`
Then instead of pass in the ir_image, you pass in the custom_ir_image into this API
`
        if (K4A_RESULT_SUCCEEDED !=
            k4a_transformation_depth_image_to_color_camera_custom(transformation,
                                                                  depth_image,
                                                                  custom_ir_image,
                                                                  transformed_depth_image,
                                                                  transformed_ir_image,
                                                                  K4A_TRANSFORMATION_INTERPOLATION_TYPE_NEAREST,
                                                                  0))
        {
            printf("Failed to compute transformed depth and custom image\n");
            return false;
        }
`

mikkamikka, 2020-04-30:
@rabbitdaxi Thank you for pointing out the right solution. However, creating a custom image from buffer with `k4a_image_create_from_buffer` method with the parameters specified in your code sample has failed. 

I've changed 'stride_bytes' parameter in the above method from `ir_image_width_pixels * (int)sizeof(uint16_t)` to `ir_image_stride_bytes`. And that solved the issue.
Now everything works as expected. Thanks again.

Here's the working method:
```
k4a_image_create_from_buffer(K4A_IMAGE_FORMAT_CUSTOM16,
                             ir_image_width_pixels,
                             ir_image_height_pixels,
                             ir_image_stride_bytes,
                             ir_image_buffer,
                             ir_image_height_pixels * ir_image_stride_bytes,
                             [](void *_buffer, void *context) {
                                  delete[](uint8_t *) _buffer;
                                  (void)context;
                             },
                             NULL,
                             &custom_ir_image)
```
cc @fractalfantasy

rabbitdaxi, 2020-04-30:
Sounds good. Feel free to close if this resolved your issue :)

closed 2020-04-30 by fractalfantasy

fractalfantasy, 2020-06-25:
@rabbitdaxi  I know this is an unconventional way of reaching you, but I think you might want to take a look at the most [active discussion](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/514) on the Azure Kinect github.

**Long story short:**  
Many Azure Kinect projects are being cancelled and put on hold because of the poor performance, accuracy and compatibility of body tracking. There's a very simple and easy to implement solution to this, which is now the [2nd most requested feature ](https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/39945454-legacy-body-tracking-like-kinect-v2) on the feedback forum, but I seems as if your collaborator **qm13** is afraid this solution would undermine his efforts - and has been promising unrealistic improvements in the current model instead.

As project leader, I think it's important you take a proper look into this - the Azure Kinect is a very important sensor for many sectors (including public health), and I think you'd agree that a major functionality like this shouldn't be held back any further.

adahbingee, 2022-05-24:
for C++ wrapper API
I leave release callback to NULL to avoid twice release
```c++
auto infer_custom = k4a::image::create_from_buffer(K4A_IMAGE_FORMAT_CUSTOM16,
                                                               infer.get_width_pixels(),
                                                               infer.get_height_pixels(),
                                                               infer.get_width_pixels() * (int) sizeof(uint16_t),
                                                               infer.get_buffer(),
                                                               infer.get_height_pixels() * infer.get_stride_bytes(),
                                                               NULL, NULL);
```