microsoft/Azure-Kinect-Sensor-SDK #1189: Infrared/Depth data broken on Intel NUC

UnaNancyOwen, 2020-04-23:
**Describe the bug**

Infrared/Depth data that using NFOV 2x2 Binned mode on Intel NUC ([NUC8i7HVK](https://www.intel.com/content/www/us/en/products/boards-kits/nuc/kits/nuc8i7hvk.html)) is broken.
The log doesn't show any warnings or errors.

[image]

It is connected to USB 3.0 port controlled by Intel's USB host controller.
It's connected using the default cable that included in Azure Kinect DK package. 

[image]

In the other mode, it is not broken.

[image]

If connect the Azure Kinect to another PC, it works fine with NFOV 2x2 Binned mode.
Therefore, Azure Kinect is fine.

What do you think?

**To Reproduce**

1. Launch the Azure Kinect Viewer.
2. Push [Open Device] button.
3. Set the Depth mode to "NFOV Binned".
4. Push [Start] button.

**Desktop (please complete the following information):**
The OS, firmware, driver and BIOS are all up to date.

 - OS with Version: Windows 1909
 - SDK Version:  v1.4.0
 - Firmware version:  AzureKinectDK_Fw_1.6.108079014.bin (included in SDK v1.4.0)

**Additional context**

wes-b, 2020-04-23:
Usually when we see this kind of issue, the device has a bad graphics driver on it. Can you look into that? We currently don't have one of these units to test with.

@rabbitdaxi FYI

UnaNancyOwen, 2020-04-24:
@wes-b Hi, Thanks for advice!
I have found that the Intel Driver Update Utility does not bring the Graphic Driver up to date. 
Therefor, I have manually updated to the latest graphic driver that donwloaded from AMD.
It is not broken data with NFOV 2x2 Binned mode. Thanks,

I'll leave a note of the driver version and URL for those who have the same environment.
NOTE: It's a WHQL version.

* Adrenalin 2020 Edition 20.2.2 Recommended (WHQL) (3/5/2020 Released)
  https://www.amd.com/en/support/graphics/radeon-rx-vega-series/radeon-rx-vega-series/radeon-rx-vega-64

closed 2020-04-24 by UnaNancyOwen