microsoft/Azure-Kinect-Sensor-SDK #1190: Using apt to install k4a package in a non-interactive environment fails

schultetwin1, 2020-04-23:
**Describe the bug**
If you attempt to install the Debian K4A package using `apt` (or `dpkg`) the install we fail by default. This is due to the interactive EULA which users need to accept (due to guidance from Microsoft) to install the Debian package.

A common scenario for this would be attempting to use `apt` to install `libk4a1.4` in a Docker container.

**To Reproduce**
1. Create a Dockerfile with a step to run `apt-get install libk4a1.4`
2. Build that docker image
3. See error -->`Sub-process /usr/bin/dpkg returned an error code (1)`

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04

schultetwin1, 2020-04-23:
This is a known issue due to the EULA we are required to present to users. We have a work around to this issue.

The EULA acceptance is done using [Debconf](https://wiki.debian.org/debconf). You can manually set entries in the debconf database to accept the EULA before installing the debian package.

The steps to do this are as follows (please substitute in the correct version of the package):

1) Set the environmental variable `DEBIAN_FRONTEND` to `noninteractive`. If you are in a Dockerfile you can do this with the `ARG` command.

   ```Docker
   ARG DEBIAN_FRONTEND=noninteractive
   ```

   This tells dpkg and debconf that you are in a non-interactive environment.

2) Manually set the hash of the accepted EULA in the debconf database.

   ```bash
   echo 'libk4a1.4 libk4a1.4/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | sudo debconf-set-selections
   ```

   NOTE: That hash may change if the EULA ever changes.

3) Manually set that the user has accepted the EULA.

   ```bash
   echo 'libk4a1.4 libk4a1.4/accept-eula boolean true' | sudo debconf-set-selections
   ```

4) Install the libk4a package. In the stdout from the install process you should see

   > License has already been accepted

schultetwin1, 2020-04-23:
We are looking at adding support for allowing users to set an environmental variable to also skip this.

This would look something like

```bash
ACCEPT_EULA=Y apt install libk4a1.5
```

Note: You'd have to be root to get this to work as I don't think `sudo` would pass the proper environmental variables.

anderwm, 2021-01-08:
This seems to work for libk4a1.x, but I don't have the same luck with libk4abt1.0.

> ENV DEBIAN_FRONTEND=noninteractive
RUN echo 'libk4a1.3 libk4a1.3/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | debconf-set-selections  
RUN echo 'libk4a1.3 libk4a1.3/accept-eula boolean true' | debconf-set-selections
RUN echo 'libk4abt1.0 libk4abt1.0/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | debconf-set-selections  
RUN echo 'libk4abt1.0 libk4abt1.0/accept-eula boolean true' | debconf-set-selections
RUN apt-get install -y --no-install-recommends k4a-tools=1.3.0 \
     libk4a1.3-dev \
     libk4abt1.0-dev

Gives me:

> 0 upgraded, 13 newly installed, 0 to remove and 193 not upgraded.
Need to get 0 B/446 MB of archives.
After this operation, 777 MB of additional disk space will be used.
Selecting previously unselected package libk4a1.3.
(Reading database ... 109924 files and directories currently installed.)
Preparing to unpack .../libk4a1.3_1.3.0_amd64.deb ...
License has already been accepted
Unpacking libk4a1.3 (1.3.0) ...
Setting up libk4a1.3 (1.3.0) ...
Selecting previously unselected package libk4abt1.0.
(Reading database ... 109935 files and directories currently installed.)
Preparing to unpack .../libk4abt1.0_1.0.0_amd64.deb ...
eula agree question could not be asked
libk4abt1.0 will not be installed
dpkg: error processing archive /var/cache/apt/archives/libk4abt1.0_1.0.0_amd64.deb (--unpack):
 new libk4abt1.0 package pre-installation script subprocess returned error exit status 2
Selecting previously unselected package libk4a1.3-dev.
Preparing to unpack .../libk4a1.3-dev_1.3.0_amd64.deb ...
Unpacking libk4a1.3-dev (1.3.0) ...
Errors were encountered while processing:
 /var/cache/apt/archives/libk4abt1.0_1.0.0_amd64.deb

Any advice?

jackjansen, 2021-04-19:
I have found a workaround for libk4abt (it needs a different hash). Here is a snippet from my `Dockerfile` that seems to work for Ubuntu 20.04 base images:

```
RUN apt-get install -y curl
# Install k4a SDK from their repository.
# Bit of a hack, see https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1263
# The simple solution by @vinesmsuic does not seem to work. This uses the manual 
# solution by @atinfinity
# Also, https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1190 is part of this solution (needed to non-interactively accept EULA)

RUN curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.3/libk4a1.3_1.3.0_amd64.deb > /tmp/libk4a1.3_1.3.0_amd64.deb
RUN curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.3-dev/libk4a1.3-dev_1.3.0_amd64.deb > /tmp/libk4a1.3-dev_1.3.0_amd64.deb
RUN curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4abt1.0/libk4abt1.0_1.0.0_amd64.deb > /tmp/libk4abt1.0_1.0.0_amd64.deb
RUN curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4abt1.0-dev/libk4abt1.0-dev_1.0.0_amd64.deb > /tmp/libk4abt1.0-dev_1.0.0_amd64.deb
RUN curl -sSL https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/k/k4a-tools/k4a-tools_1.3.0_amd64.deb > /tmp/k4a-tools_1.3.0_amd64.deb
RUN echo 'libk4a1.3 libk4a1.3/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | debconf-set-selections
RUN echo 'libk4abt1.0	libk4abt1.0/accepted-eula-hash	string	03a13b63730639eeb6626d24fd45cf25131ee8e8e0df3f1b63f552269b176e38' | debconf-set-selections
RUN dpkg -i /tmp/libk4a1.3_1.3.0_amd64.deb
RUN dpkg -i /tmp/libk4a1.3-dev_1.3.0_amd64.deb
RUN dpkg -i /tmp/libk4abt1.0_1.0.0_amd64.deb
RUN dpkg -i /tmp/libk4abt1.0-dev_1.0.0_amd64.deb
RUN apt-get install -y libsoundio1
RUN dpkg -i /tmp/k4a-tools_1.3.0_amd64.deb
```

apockill, 2021-11-30:
If I might add, I was able to also use the `yes` command in conjunction with `DEBIAN_FRONTEND` to get an automated workflow. Here's what worked for me:

```
yes yes | DEBIAN_FRONTEND=dialog apt-get install -y /tmp/*.deb
```

earlev4, 2021-12-23:
Thank you very much @jackjansen!!! 
Your comment helped me solve my issue with the build of my Dockerfile.

Looks like there are newer .debs available:
```
libk4a1.3_1.3.0_amd64.deb > libk4a1.4_1.4.1_amd64.deb
libk4a1.3-dev_1.3.0_amd64.deb > libk4a1.4-dev_1.4.1_amd64.deb
libk4abt1.0_1.0.0_amd64.deb > libk4abt1.1_1.1.0_amd64.deb
libk4abt1.0-dev_1.0.0_amd64.deb > libk4abt1.1-dev_1.1.0_amd64.deb
k4a-tools_1.3.0_amd64.deb > k4a-tools_1.4.1_amd64.deb
```

How did you obtain the following EULA hash strings? 

```
RUN echo 'libk4a1.3 libk4a1.3/accepted-eula-hash string 0f5d5c5de396e4fee4c0753a21fee0c1ed726cf0316204edda484f08cb266d76' | debconf-set-selections
RUN echo 'libk4abt1.0	libk4abt1.0/accepted-eula-hash	string	03a13b63730639eeb6626d24fd45cf25131ee8e8e0df3f1b63f552269b176e38' | debconf-set-selections
```

I would like to obtain new EULA hash strings for the new .debs.

Thanks again! Greatly appreciate your comments.

yukilab2, 2022-02-28:
> How did you obtain the following EULA hash strings
Once you manually accept EULA, you can see those hash by typing `sudo debconf-show libk4a1.3`, `sudo debconf-show libk4abt1.0`

yzqin, 2022-07-01:
> Thank you very much @jackjansen!!! Your comment helped me solve my issue with the build of my Dockerfile.
>
> Looks like there are newer .debs available:
>
> ```
> libk4a1.3_1.3.0_amd64.deb > [...]

Thanks for the solutions. I found that for different version of libk4a and libk4abt the EULA hash strings is exactly the same so we can directly use the hash strings by @jackjansen.

k1915361, 2023-11-10:
> I have found a workaround for libk4abt (it needs a different hash). Here is a snippet from my `Dockerfile` that seems to work for Ubuntu 20.04 base images:
>
> ```
> RUN apt-get install -y curl
> # Install [...]

You may have to install these dependencies first.  
`RUN apt-get -y install python3.9-dev libgl1 libxrandr2 libxinerama1 libxcursor1`