microsoft/Azure-Kinect-Sensor-SDK #1192: Could I use the Azure Kinect DK for a motion deformation warning system?

ohynkwn, 2020-04-27:
**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

ohynkwn, 2020-04-27:
I know that Azure Kinect DK is such a great device to motion sensing and body tracking.

Therefore, I would like to apply this machine to detect motion as a difference from standard (or original) motion range. For instance, setting a human-size box and tracking someone's motion to confirm they are fitted in the box or not. Furthermore, I want to link with the warning system to alarm for the user about occurring some motion differential situations.

Consequently, I ask all the experienced developers. Could I use the Azure Kinect DK for a motion deformation warning system? Thank you for reading this.

qm13, 2020-04-30:
@ohynkwn we are aware of a number of customers who are using Azure Kinect and custom AI or body tracking in this way.

closed 2020-05-21 by qm13