microsoft/Azure-Kinect-Sensor-SDK #1199: Multiple cameras instances raise replace_sample(). capturesync_drop, releasing capture early due to full queue

carcormir, 2020-05-01:
Hi,

**Describe the bug**
After running the three instances of K4Viewer for a few minutes I get the following error (see log file below from the camera that stops working, the other two stay active)

This is the error:
`[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS: 326682755 type:Depth`

It seems like (from the log file) my GPU is not working hard enough. However my hardware consists on 2 GPUs GeForce RTX 2080 Ti/PCIe/SSE2, which I assume should do the job of keeping the processing under 33ms as mentioned in #1112 
It seems like when I check the activity of the GPU being used, there is a 64mb upper limit from the SDK per k4aviewer instance that might not be sufficient to keep up with the synchronization when having multiple cameras running.

**To Reproduce**
Open three instances of k4aviewer and select BGRA format, 30fps, 2160p and the rest of values should be the default ones.

**Expected behavior**
I expect that the GPUs are used as expected.

Logs
[test_1_camera_k4a.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4564465/test_1_camera_k4a.log)

**Desktop:**
Hardware
CPU: AMD® Ryzen threadripper 2970wx 24-core processor × 48
2 GPUs: GeForce RTX 2080 Ti/PCIe/SSE2
Motherboard: ASRock x399 Taichi
Ram: 64Gb

Software
Ubuntu 18.04
SDK: Built from source. Latest develop commit available. v1.4.1
I also have libk4a1.3 and libk4a1.3-dev installed from Debian packages but k4aviewer is run from the built from source version
Firmware version: 1.6.102 (RGB) 1.6.75 (Depth)

Thanks in advance for the help!

wes-b, 2020-05-01:
I see a couple things in the log.

> [ warning  ] : is_fw_version_compatable(). Firmware version for RGB is 1.6.102. Consider upgrading to 1.6.108 or newer.
[ warning  ] : is_fw_version_compatable(). Firmware version for Depth is 1.6.75. Consider upgrading to 1.6.79 or newer.

You need to upgrade your firmware, this version is fairly old. Grab 1.6.110079014 from the 1.4.1.Alpha package on [Nuget](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.1-alpha.0). That firmware is good and expected to be officially released soon.

>[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 45ms (this may be transient). 

Its hard to tell from this version of the log if this is a 1 second session or several minutes. I do expect to see this warning occasionally when the PC is active and when a stream starts up. For future logs please log to a file instead of copy and pasting from the viewer, that will contain event timestamps to better understand timing. `K4A_ENABLE_LOG_TO_A_FILE=k4a_1.log` from https://docs.microsoft.com/en-us/azure/Kinect-dk/troubleshooting. You will need to use 3 different log files names for the 3 different viewer instances so they don't conflict with each other. You will see critical error messages in the viewer about firmware version, it is safe to ignore it. BTW, this is the only messages that indicates GPU activity. So no indication GPU is under utilized. And I don't having a 2nd GPU will help. I don't believe we load balance to other GPUs; @rabbitdaxi?

> [ warning  ] : DecodeMJPEGtoBGRA32(). MJPEG decode failed, dropping image: -1

I see more of these than expected. I do notice that this happens occasionally when a streaming session is started or stopped. Is it possible the other sessions were stopped and started a bunch during this time? The log file with timestamps would help here too.

> [ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS: 326749422 type:Depth

This message looks like it starts happening continuously. It indicates the color camera stopped streaming for an unknown reason. Can you repro with an "Info" level log? `K4A_LOG_LEVEL=i`

rabbitdaxi, 2020-05-01:
@wes-b > And I don't having a 2nd GPU will help. I don't believe we load balance to other GPUs; @rabbitdaxi? > yup, we do not load balance to other GPUS, 1 GPU is normally good enough for us to reach to real time even just some integrated GPU. If user has multiple GPU, user can use software from the GPU vendor to choose the default GPU for the program.

carcormir, 2020-05-06:
Hi @wes-b,
Sorry for the delay.

I have updated the firmware of the three cameras to the latest officially released version available.

I have performed the same test and collected the log files at info level as you suggested. As you may see below, k4a_2.log is the one that fails.
Test: Open three independent instances of k4aviewer with BGRA format, 15fps, 2160p and WFOV Unbinned.

Log files:
[Camera 1: k4a_1.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4585854/k4a_1.log)
[Camera 2: k4a_2.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4585855/k4a_2.log)
[Camera 3: k4a_3.log](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4585856/k4a_3.log)

I hope this helps somehow.
Thanks!

wes-b, 2020-05-08:
In the logs I see all 3 k4aviewer session start around 10:42:27 and then they run for several minutes before k4a_2.log hits the error. During this time I see several of the “DecodeMJPEGtoBGRA32(). MJPEG decode failed, dropping image: -1" errors. In all my testing I could only trigger these errors while starting / stopping other camera’s. I am assuming you were not using other USB devices during this time. 

Even with those errors, there is ~10s  after that and before our first indicator that color images stopped arriving “capturesync_drop, releasing capture early due to full queue TS: 191546111 type:Depth". Unfortunately we don’t know why they stopped arriving. And this is ultimately the reason for the error.

Sorry I don’t have anything solid. I suspect you are having troubles if your USB card. If I had access to your machine I would try again using the native USB host controller built into your motherboard. My Linux PC, which is OLD, is running this same scenario for days with no repo. It can only do 5-10 FPS for each camera, but it runs error free.

Another idea, is make sure the timeouts you are using are large. K4aViewer is too small at 2s. On my setup I did see a period of time where the sdk was “releasing capture early", but the PC recovered, and as a result no error was reported. I was fortunate that period of time was less than the timeout, you might not have been.

carcormir, 2020-05-13:
Hi @wes-b,

>  I am assuming you were not using other USB devices during this time.

I was not using anything else just the three cameras connected to the USB card that has 3 independent controllers (not the one in the motherboard)

> I would try again using the native USB host controller built into your motherboard

I run the same test connecting the three cameras to the built in USB controller and it seems to be working. Some frames are dropped sometimes but no error after running 3 instances of k4aviewer for more than 2 hours. (This test was performed two times)

Notice that I am using the latest version of the SDK available in the develop branch and, for the depth engine library I installed libk4a1.3. When I tried using libk4a1.4, the built from source version of the SDK was not able to find the depth engine and would raise an error that would not let me open the devices.

We are looking forward for the official release so we can finally try together with the Azure Kinect ROS driver. Any place to track the release date for the .deb package?

You may close this thread and I will open a new if needed.

Thanks for the help!

wes-b, 2020-05-13:
Glad to hear its working better!

>Any place to track the release date for the .deb package?

No official date, but should be soon. Keep an eye on https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1206, it will be updated when the release is done.

closed 2020-05-13 by wes-b

christiandreher, 2020-09-15:
@carcormir May I ask what motherboard you are using? I experience severe errors with my Azure Kinects and several extention cords, I can barely use them. I also have 2x 2080 RTX Ti, so this is not the issue. However I think that my mainboard might be the cause... I have no issue at all when I operate the Azure Kinects with the very same cables on my Notebook.