microsoft/Azure-Kinect-Sensor-SDK #1201: Imperfect alignment in IR&depth to Color transformation

SeanStephensenAIRM, 2020-05-01:
I'm trying to transform depth and IR images to the same reference frame as the RGB image. The alignment is close but noticeably off. Here is a synchronized capture of color, depth, IR images followed by an overlayed depth/color comparison and IR/color comparison. By looking at the white metal rack, it is clear that the alignment is off (appears to be the same misalignment between the IR&depth images). After the images is my code for depth --> color and IR --> color. Any advice on fixing this issue would be appreciated. 

[image]

[image]

[image]

[image]

[image]

```
k4a_image_t depth_to_color(CAMERA camera, k4a_image_t depth_image, k4a_image_t color_image)
{
	//attributes of the color image will be required
	uint color_height = k4a_image_get_height_pixels(color_image);
	uint color_width = k4a_image_get_width_pixels(color_image);
	uint color_stride = k4a_image_get_stride_bytes(color_image);
	//create a transformed depth image to be written into (Must be same format as depth_image, and same dimensions as color_image)
	k4a_image_t transformed_depth_image = NULL;
	switch (k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16, color_width, color_height, color_width * (int)sizeof(uint16_t), &transformed_depth_image))
	{
	case K4A_RESULT_SUCCEEDED:
		break;
	case K4A_RESULT_FAILED:
		printf("Blank depth image creation failed (depth_to_color) :(\n");
	}
	//apply the transformation
	switch (k4a_transformation_depth_image_to_color_camera(camera.intr_transform, depth_image, transformed_depth_image))
	{
	case K4A_RESULT_SUCCEEDED:
		break;
	case K4A_RESULT_FAILED:
		printf("Depth Transformation failed :(\n");
	}
	return transformed_depth_image;
}

k4a_image_t IR_to_color(CAMERA camera, k4a_image_t depth_image, k4a_image_t IR_image_raw, k4a_image_t color_image)
{
	//first we must transform the raw depth image into the same reference frame as the color image, so they may be overlayed
	//attributes of the color image will be required
	uint color_height = k4a_image_get_height_pixels(color_image);
	uint color_width = k4a_image_get_width_pixels(color_image);
	uint color_stride = k4a_image_get_stride_bytes(color_image);
	//create a transformed image to be written into (Must be same format as depth_image, and same dimensions as color_image)
	//first, the IR_image must be changed into CUSTOM16 format (captured in IR16 format)
	k4a_image_t IR_image = change_IR_format(IR_image_raw);
	//now create blank image containers
	k4a_image_t transformed_IR_image = NULL;
	k4a_image_t transformed_depth_image = NULL;
	switch (k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM16, color_width, color_height, color_width * (int)sizeof(uint16_t), &transformed_IR_image))
	{
	case K4A_RESULT_SUCCEEDED:
		break;
	case K4A_RESULT_FAILED:
		printf("Blank IR image creation failed (IR_to_color) :(\n");
	}
	switch (k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16, color_width, color_height, color_width * (int)sizeof(uint16_t), &transformed_depth_image))
	{
	case K4A_RESULT_SUCCEEDED:
		break;
	case K4A_RESULT_FAILED:
		printf("Blank depth image creation failed (IR_to_color) :(\n");
	}
	//apply the transformation
	switch (k4a_transformation_depth_image_to_color_camera_custom(camera.intr_transform, depth_image, IR_image, transformed_depth_image,
		transformed_IR_image, K4A_TRANSFORMATION_INTERPOLATION_TYPE_NEAREST, 0))
	{
	case K4A_RESULT_SUCCEEDED:
		break;
	case K4A_RESULT_FAILED:
		printf("IR/depth Transformation failed :(\n");
	}
	return transformed_IR_image;
}
```

qm13, 2020-05-06:
@SeanStephensenAIRM it is hard to tell from these images what you are experiencing. Some thoughts are:
1.	The images misalignment is after transform the depth/IR to color domain which should use the camera calibration along with the depth image
2)	The misalignment is quite small (I estimate a few RGB pixels). This could be attributed to:
a.	it could be just the calibration vs hardware is slightly different
b.	it could be the depth error itself leading to this small diff.

2b in particular could be a result of multipath in the scene. Nevertheless the expected bias can be up to 11mm +-0.1% of distance.

Could you repeat the experiment with a flat markerboard and in an open space?

SeanStephensenAIRM, 2020-05-15:
@qm13 Thank you for your response and sorry for the delay in mine. Here's a rectangular box placed on the carpet - I haven't included the depth image because it's harder to tell (box is too close to the carpet), but here's the IR and color images shown side by side with clear misalignment at the box edges. I can take some better pictures on Tuesday that would have more useful depth images to compare as well. Anything more specific you'd like to see in the next round of images?

[image]

tim-depthkit, 2020-05-21:
Seems like this is a duplicate of #1058 

Very helpful to see the comparison images like this. @SeanStephensenAIRM Do you have the source available to produce these images? Could be the start a nice diagnostic utility if combined with something like OpenCV findChessboards and then compute the error between the two resulting point sets.

SeanStephensenAIRM, 2020-05-22:
I agree, this does seem to be the same issue - feel free to merge if possible.  I have also noticed many instances where this error is manifested in point clouds. I can't share my full source since it is IP of the company I work for, but the process is just taking a simple capture as in many example Kinect codes, applying the transformations that I was able to provide in my original post, and then using OpenCV to imwrite each of the files out. If I have time next week I can try some OpenCV chessboard captures with my code where it would report the error between corners.

qm13, 2020-06-12:
We believe this is a hardware issue. Can you please report the issue [here](https://support.microsoft.com/en-us/supportforbusiness/productselection?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813))

SeanStephensenAIRM, 2020-06-15:
@qm13 I think I have done what you're asking? Honestly not confident that I found the right spot that you wanted me to report the issue. Can you confirm that [this](https://docs.microsoft.com/en-us/answers/questions/36212/image-misalignment-from-azure-kinect-camera.html) is what you wanted?

qm13, 2020-06-15:
@SeanStephensenAIRM did you use the link I provided in my comment. Full link is:

https://support.microsoft.com/en-us/supportforbusiness/productselection?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813

SeanStephensenAIRM, 2020-06-15:
> @SeanStephensenAIRM did you use the link I provided in my comment. Full link is:

Thanks, this new link works. Your old link took me to a very general Microsoft support page from which I had to manually navigate to my best guess at somewhere to file an issue. Your new link is specific to a direct support form which I've now filled out.

qm13, 2020-06-15:
Weird. When I saw your original response the first thing I thought was I had given you the wrong link. I checked it and it works for me. Anyway glad you have now completed the support form. You will also get a response on the original incident you filed. The incident team knows I am covering the issue directly.

closed 2020-07-11 by qm13