microsoft/Azure-Kinect-Sensor-SDK #1202: Kinect recorder rendering black IR and Depth frames

fractalfantasy, 2020-05-02:
Trying to render out some video of depth data with k4arecorder.exe, and even when not in passive IR mode, the video of IR and Depth are turning out black/extremely dark.

Attached output data below:
https://www.dropbox.com/s/cwdcv3526jvi9ex/kinect%20viewer.PNG?dl=0

**To Reproduce**
1. Record 3 seconds of data in K4arecorder.exe, in command prompt type:
"C:\Program Files\Azure Kinect SDK v1.4.0\tools\k4arecorder.exe" -d WFOV_2X2BINNED --record-length 3 Desktop/output.mkv
2. parse into 3 videos using ffmpeg:
"ffmpeg -i C:/Users/FF/Desktop/output.mkv -map 0:2 -vsync 0 C:/Users/FF/Desktop/ir.mp4 & ffmpeg -i C:/Users/FF/Desktop/output.mkv -map 0:1 -vsync 0 C:/Users/FF/Desktop/depth.mp4 & ffmpeg -i C:/Users/FF/Desktop/output.mkv -map 0:0 -vsync 0 C:/Users/FF/Desktop/color.mp4"

**Expected behavior**

Depth data should be visible and IR should be active like in Azure Kinect Viewer:

**Screenshots**
parsed color:
[image]
parsed IR:
[image]
parsed Depth:
[image]

Expected:
[image]

wes-b, 2020-05-04:
K4aViewer applies a grey scale to the IR stream and the color hue to the depth camera stream. Neither of those are applied to the raw recorded data. You would need to do something similar in your own stream.

fractalfantasy, 2020-05-04:
Ok thanks, I’ll try to find a way to process those streams with ffmpeg and report back if I have any luck.

Feel free to close this topic.

closed 2020-05-04 by wes-b

krusic22, 2023-06-14:
I assume it didn't work out?
I'm trying to record the depth camera using ffmpeg and the image is just too dark.
I probably tried every filter under the sun, but none of them really help out.