microsoft/Azure-Kinect-Sensor-SDK #1206: Publish SDK 1.4.1

wes-b, 2020-05-06:
**MSI Link:** Comming Soon
**Nuget Link:** [[1.4.1-beta.0](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.1-beta.0)]
**Deb package link:** [AMD64 Comming Soon] [ARM64 Comming Soon]
**Firmware:** [1.6.110079014]
Summary of changes [are here](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/release/1.4.x/CHANGELOG.md)

Kristian-Gregersen-KRGR, 2020-05-20:
Hi!
When do you expect to release the Debian package? Is “soon” this week, next week or...? 
Thanks!

UnaNancyOwen, 2020-06-26:
@wes-b Azure Kinect Sensor SDK v1.4.1 seems has been published at [NuGet Gallery](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.1) and [Package Repository](https://packages.microsoft.com/ubuntu/18.04/multiarch/prod/pool/main/libk/libk4a1.4/). At the least, that NuGet package does not appear to be a beta version. However, Azure Kinect Sensor SDK v1.4.1 is not tagged at [GitHub](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/releases).  What’s happening?

wes-b, 2020-06-26:
Thank you for pointing this out. There was a failure in our automation. We will get the MSI and other updates out shortly.

UnaNancyOwen, 2020-07-03:
@wes-b Thanks, Please also update download link in Microsoft Docs. :)
https://docs.microsoft.com/en-us/azure/kinect-dk/sensor-sdk-download#windows-download-link

closed 2020-07-10 by wes-b

chethanab16, 2020-09-10:
Hi, 
1. Is Azure-Kinect-Sensor-SDK supported on Lattepanda (Windows OS) ,  Raspberry Pi 4 (Windows ARM OS) and Intel Compute Stick ? 
2. Does Microsoft hololense  azure kinect  is same as new azure kinect camera?
thank you in advance.