microsoft/Azure-Kinect-Sensor-SDK #1207: [feature request] Add C# wrapper support for Linux

UnaNancyOwen, 2020-05-07:
**Is your feature request related to a problem? Please describe.**

C# wrapper of Azure Kinect Sensor SDK (and Azure Kinect Body Tracking SDK) is not provided for Linux.
Please consider to provide C# wrapper for Linux (x86-64, arm64). Thanks,

**Describe the solution you'd like**

Currently, C# wrapper generate by build Visual Studio project (<code>src/csharp/K4a.sln</code>).
I think it is necessary to be changed to generate by build files using CMake for closs platform.
Then, Please add C# wrapper for Linux to NuGet package.

dbruning, 2020-05-25:
Any thoughts on priority or timeline for this? It’s a blocker for my project.

qm13, 2022-05-25:
Added in v1.1.2