microsoft/Azure-Kinect-Sensor-SDK #1208: Keep getting Pixels not aligned to stride of each line

cinis77, 2020-05-07:
Hi there,

I am getting Microsoft.Azure.Kinect.Sensor.AzureKinectException: 'Pixels not aligned to stride of each line' exception when i try to call .Color.GetPixels() method. 

[image]

To Help code:
private void TryGetDepath()
        {
            var device = Device.Open();
            device.StartCameras(new DeviceConfiguration
            {
                CameraFPS = FPS.FPS5,
                ColorResolution = ColorResolution.R1080p,
                DepthMode = DepthMode.WFOV_Unbinned,
                WiredSyncMode = WiredSyncMode.Standalone,
                SynchronizedImagesOnly = true
            });
            var calibration = device.GetCalibration();
            var transformation = calibration.CreateTransformation();
            Capture capture = device.GetCapture();

            while (capture == null || capture.Color == null || capture.Depth == null)
            {
                capture = device.GetCapture();
            }

            var colorPixel = capture.Color.GetPixels<abc>();
            Console.WriteLine(colorPixel.Length);
            foreach (var c in colorPixel.Span)
            {
                Console.WriteLine($"{c.a} {c.r} {c.g} {c.b}");
               
            }

            var depthFromColor = transformation.ColorImageToDepthCamera(capture);
            var depthFromColorPixel = depthFromColor.GetPixels<abc>();
            Console.WriteLine(depthFromColorPixel.Length);
        
            foreach (var c in depthFromColorPixel.Span)
            {
                Console.WriteLine($"{c.a} {c.r} {c.g} {c.b}");
           
                
            }

            var depthPixel = capture.Depth.GetPixels<UInt16>();
            Console.WriteLine(depthPixel.Length);
       

            foreach (var d in depthPixel.Span)
            {
                if (d > 0)
                {
                    Console.WriteLine($"{d} ");
                  
                }
            
            }

        }

    struct abc
    {
        public byte a;
        public byte r;
        public byte g;
        public byte b;
    }

wes-b, 2020-05-07:
What image format are you using? MJPEG doesn't have stride.

cinis77, 2020-05-08:
Found the issue thanks

closed 2020-05-08 by cinis77

Fenikkel, 2021-08-19:
> 
> 
> Found the issue thanks

Hi, can you share the issue solution? Thanks

Alessandro-Minerba, 2021-12-23:
Hi, can someone share the solution to this issue? I understood it's quite common.

Capricoornus, 2022-04-09:
Hi there. I had same problem. 
I returned **depth** image from method which should return **color** image. Simple misswrote which lead to problem that I then tried to pass returned image into BGRA array which led to _"Pixels not aligned to stride of each line error"._