microsoft/Azure-Kinect-Sensor-SDK #1209: No Instruction on using OpenCV with Azure Kinect Camera

tahrir94, 2020-05-08:
I need to detect any objects within a region of interest in real-time, other than two that will always be present, using the Kinect camera for a project. It can be a simple boxing of the objects to get a count, but I'm not being able to find any resources to help with this. OpenCV can do this easily but after much searching I wasn't able to find any guidance on using opencv library with the Kinect camera. Any help regarding this will be greatly appreciated!

UnaNancyOwen, 2020-05-08:
I hope that my repository will help you.
This is sample program for Azure Kinect with OpenCV.
https://github.com/UnaNancyOwen/AzureKinectSample

tahrir94, 2020-05-09:
@UnaNancyOwen Thank you so much! I'll try to implement this but if I get stuck with any unforeseen errors, may I seek your help? If yes, please let me know the best way to reach you :)

UnaNancyOwen, 2020-05-09:
If that's problem about Azure Kinect Sensor SDK or Sensor, You should write issues of this repository. 
Otherwise, You can contact me at issues of my repository or by email.
(Please close this issue when this is resolved.)

tahrir94, 2020-05-10:
Hi Tsukasa,

Thank you so much for agreeing to help. As you can see in the attachment,
I'm getting an error due to cmake not being able to find opencv when I try
to generate the binaries. I've copied my download of opencv into the folder
containing CMakeLists.txt in your project, as you can see in the second
attachment, but that didn't work either. I've been trying to solve this
since yesterday with no luck so your help will mean a lot. I may face the
same issue with k4a so if you know where these libraries should be kept for
cmake please let me know :)

Very gratefully,
Tahrir

tahrir94, 2020-05-10:
Hi Tsukasa,

Thank you so much for agreeing to help. As you can see in the attachment,
I'm getting an error due to cmake not being able to find opencv when I try
to generate the binaries. I've copied my download of opencv into the folder
containing CMakeLists.txt in your project, as you can see in the second
attachment, but that didn't work either. I've been trying to solve this
since yesterday with no luck so your help will mean a lot. I may face the
same issue with k4a so if you know where these libraries should be kept for
cmake please let me know :)

Very gratefully,
Tahrir
[image]
[image]

UnaNancyOwen, 2020-05-10:
You should be to learn how to using CMake before you start it.
Then, It is an issue that has nothing to do with Azure Kinect.
I don't think that is a discussion should be having here.
Please close this issue.

tahrir94, 2020-05-10:
Ok, I've been learning how to use OpenCV, sorry if the query was irrelevant. If I face issues that are specific to your repo I will reach out. Thank you so much for your help, I'm closing this issue.

closed 2020-05-10 by tahrir94