microsoft/Azure-Kinect-Sensor-SDK #1212: k4a_DIR not found when configuring in cmake

tahrir94, 2020-05-11:
Hello, I am trying to build my visual studio project with cmake but I'm not being able to get past the following error when I configure it in cmake:

Could not find a package configuration file provided by "k4a" with any of
  the following names:
    k4aConfig.cmake
    k4a-config.cmake
  Add the installation prefix of "k4a" to CMAKE_PREFIX_PATH or set "k4a_DIR"
  to a directory containing one of the above files.  If "k4a" provides a
  separate development package or SDK, be sure it has been installed.

As you can see in the attached CMakeLists.txt file, I've set the k4a_DIR path to the cmake folder where I installed this repo. However, I see that the folder has a k4aConfig.cmake,ini instead of just cmake and I think that may be causing the error. Hopelessly stuck with this so I would REALLY appreciate some help
[image]
!

wes-b, 2020-05-11:
This is not a lot of information, but we do have a crumb of info here about using VC2017: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md#building-using-visual-studio. 

If that doesn't help, then can you share repro steps so we can reproduce the problem?

tahrir94, 2020-05-14:
@wes-b you're right, that's not enough information. Here are my retro steps:
1) Cloned this repo for using OpenCV with Azure Kinect: https://github.com/UnaNancyOwen/AzureKinectSample
2) Followed this tutorial to install OpenCV: https://www.youtube.com/watch?v=_fqpYLM6SCw
3) cloned this Azure Kinect SDK repo
3) In the project folder for point_cloud (..\AzureKinectSample\sample\c\point_cloud), edited CMakeLists.txt to set paths to the install locations of opencv and azurekinect sdk (lines 12 and 13 in the snippet above). This fixed the same error for opencv but as I already said, this repo does not contain a k4aConfig.cmake so k4a cannot be configured.

Please let me know if you're able to find any solutions after reproducing the problem. Feel free toask further questions. Thanks for your help!

ZJU-PLP, 2020-07-01:
> @wes-b you're right, that's not enough information. Here are my retro steps:
> 
> 1. Cloned this repo for using OpenCV with Azure Kinect: https://github.com/UnaNancyOwen/AzureKinectSample
> 2. Followed this tutorial to install OpenCV: https://www.youtube.com/watch?v=_fqpYLM6SCw
> 3. cloned this Azure Kinect SDK repo
> 4. In the project folder for point_cloud (..\AzureKinectSample\sample\c\point_cloud), edited CMakeLists.txt to set paths to the install locations of opencv and azurekinect sdk (lines 12 and 13 in the snippet above). This fixed the same error for opencv but as I already said, this repo does not contain a k4aConfig.cmake so k4a cannot be configured.
> 
> Please let me know if you're able to find any solutions after reproducing the problem. Feel free toask further questions. Thanks for your help!
Hello boy! I also meet the problem. Could you mind sharing your solving method in terms of this problem ?

ZJU-PLP, 2020-07-01:
> @wes-b you're right, that's not enough information. Here are my retro steps:
>
> 1. Cloned this repo for using OpenCV with Azure Kinect: https://github.com/UnaNancyOwen/AzureKinectSample
> 2. Followed [...]

Hello boy! I also meet the problem. Could you mind sharing your solving method in terms of this problem ?

wes-b, 2020-07-01:
Sorry I just realized this is not in the samples repo. This is @UnaNancyOwen sample; can you help please?

UnaNancyOwen, 2020-07-01:
The Azure Kinect Sensor SDK installer doesn't include the files to find library using CMake. Therefor, You can't find Azure Kinect Sensor SDK using CMake. If you want to find Azure Kinect Sensor SDK using CMake, You need to build and install Azure Kinect Sensor SDK from source code yourself with CMake. It is usually installed under "C:\Program Files\K4A" by build INSTALL project of Azure Kinect Sensor SDK. (If you newer about CMake, Please learn how to use CMake basicaly in first. It will be helpful for you.)

```bat
rem please run command prompt with administrater.
rem please set Git and CMake path to environment variable PATH.

cd C:\
git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git
cd Azure-Kinect-Sensor-SDK
git checkout -b v1.4.1 refs/tags/v1.4.1
mkdir build && cd build
cmake .. -G "Visual Studio 16 2019" -A x64 -DCMAKE_INSTALL_PREFIX="C:/Program Files/K4A"
cmake --build . --config Release
cmake --build . --target install --config Release
```

[image]

@wes-b Please consider that including CMake files in the installer again in your team. It is useful for all developers. (e.g. OpenCV has CMake files in pre-built packages.)

jaypee37, 2020-07-10:
@tahrir94  You can get the nuget package from [here](https://www.nuget.org/packages/microsoft.azure.kinect.sensor/)

Then you can try linking against the k4a libs from the nuget package. I have attached a example cmake file here. you may need to change the path to the k4a binaries based on your build type.

[image]

tobibaum, 2021-06-25:
@UnaNancyOwen thanks for the instructions! these are very helpful also for your repo.

I have a little problem: 
the K4A folder in Program Files is not created. Any ideas why that could be?

thanks!

lhamza98, 2021-07-26:
@tobibaum 
I seem to be having the same problem with the K4A folder not being created. Did you end up finding a solution?

tobibaum, 2021-07-26:
I ended up switching my development over to ubuntu instead, where I was more comfortable with linking and such.
here is a cmake that worked for me:

lhamza98, 2021-07-26:
@tobibaum
Okay thank you!