microsoft/Azure-Kinect-Sensor-SDK #1213: c++ Refernce  device.getcapture() not listed

Ayala65, 2020-05-11:
while (!stop && !visualization.wasStopped())
    {
        switch(device.get_capture())
        {
        case K4A_WAIT_RESULT_SUCCEEDED:
            break;
        case K4A_WAIT_RESULT_TIMEOUT:
            printf("Timed out waiting for a capture\n");
            continue;
            break;
        case K4A_WAIT_RESULT_FAILED:
            printf("Failed to read a capture\n");
            device.close();
            return 1;
        }

    }

E0304	no instance of overloaded function "k4a::device::get_capture" matches the argument list	te

wes-b, 2020-05-11:
Sorry, I don't understand your issue. Please clearly tell us what the issue is, what you expect to happen, and steps to reproduce the problem

Ayala65, 2020-05-12:
I try to use the method get_capture from device to obtain frame per frame of the depth camera but when i write this "device.get_capture()" the error is launched even when i try in a new program  the same error is launched.

wes-b, 2020-05-12:
Can you share your code?

Ayala65, 2020-05-12:
#pragma warning(disable : 4996)
// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

#include <stdio.h>
#include <fstream>
#include <sstream>
#include <vector>
#include <algorithm>
#include<k4a/k4a.hpp>
#include <math.h>

using namespace std;

// Enable HAVE_OPENCV macro after you installed opencv and opencv contrib modules (kinfu, viz), please refer to README.md
#define HAVE_OPENCV
#ifdef HAVE_OPENCV
#include <opencv2/core.hpp>
#include <opencv2/calib3d.hpp>
#include <opencv2/opencv.hpp>
#include <opencv2/rgbd.hpp>
#include <opencv2/viz.hpp>
using namespace cv;
#endif

#ifdef HAVE_OPENCV
void initialize_kinfu_params(kinfu::Params& params,
    const int width,
    const int height,
    const float fx,
    const float fy,
    const float cx,
    const float cy)
{
    const Matx33f camera_matrix = Matx33f(fx, 0.0f, cx, 0.0f, fy, cy, 0.0f, 0.0f, 1.0f);
    params.frameSize = Size(width, height);
    params.intr = camera_matrix;
    params.depthFactor = 1000.0f;
}

template<typename T> Mat create_mat_from_buffer(T* data, int width, int height, int channels = 1)
{
    Mat mat(height, width, CV_MAKETYPE(DataType<T>::type, channels));
    memcpy(mat.data, data, width * height * channels * sizeof(T));
    return mat;
}
#endif

#define INVALID INT32_MIN
typedef struct _pinhole_t
{
    float px;
    float py;
    float fx;
    float fy;

    int width;
    int height;
} pinhole_t;

typedef struct _coordinate_t
{
    int x;
    int y;
    float weight[4];
} coordinate_t;

typedef enum
{
    INTERPOLATION_NEARESTNEIGHBOR, /**< Nearest neighbor interpolation */
[... 513 lines cut ...]
                << out_points.at<float>(i, 2) << " "
                << out_normals.at<float>(i, 0) << " "
                << out_normals.at<float>(i, 1) << " "
                << out_normals.at<float>(i, 2) << endl;
        }
        ofstream ofs_text(output_file_name, ios::out | ios::app);
        ofs_text.write(ss.str().c_str(), (streamsize)ss.str().length());
    }
    else if (key == 'q')
    {
        stop = true;
    }

    depth_image.reset();
    undistorted_depth_image.reset();
    capture.reset();
    
    }
    lut.reset();
    cv::destroyAllWindows();
#endif
    device.close();

    return 0;
}

Ayala65, 2020-05-12:
[test.zip](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/4618267/test.zip)

wes-b, 2020-05-13:
I see you are using our kinfu sample. I suggest confirming you can get that sample to work as published have you tried that.

Ayala65, 2020-05-13:
Yes, i have compiled that example and im using it to make another program but when i tried the c++ reference i have this problem so i remake that sample to c++, i have the same problem with getcapture