microsoft/Azure-Kinect-Sensor-SDK #1214: External Sync Not Working

ETLang, 2020-05-11:
Hi, I'm trying to get external sync working between two Kinects. I'm using SDK version 1.2.0 and the Viewer app it comes with on Windows 10. I've tried updating the firmware, running the Kinects on different PCs, and on the same PC. I've tried starting the master first, and starting the sub first. I've tested the cable to ensure it properly carries audio data, so it's not broken. I've been following the examples in the documentation religiously. I've ensured that the sub detects a "sync in" connection, and that the master detects a "sync out" connection. Everything seems to be in order.

Nevertheless, the sub Kinect never receives any signal from the master, so never captures any frames at all. It's just black. Any thoughts on what the cause might be?

Thanks,
Evan

ETLang, 2020-05-11:
I've also tried switching the role of the Kinects, in case one of the sync ports was bad. Still, no luck.

wes-b, 2020-05-14:
Have you reviewed this topic? https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync

With K4aViewer, start the subordinate first, then start the master.

The fact you never saw captures on the subordinate indicates an issue with the cable or connection jacks. Try swapping master and subordinate, or use another cable.

ETLang, 2020-05-14:
Yes, I've based my steps mostly on that doc. I've tried everything you mentioned except getting a different cable. I don't have a different cable right now, so I suppose I'll have to order one.

Is there a particular one you can recommend?

wes-b, 2020-05-14:
This sounds like a hardware issue. If the k4aViewer works fine standalone on both devices, and then the subordinate fails to get captures when the Sync In is connected to the Sync Out jack, then likely one of the Kinects is bad. Do you have a 3rd device to test with? I would consider buying a 3rd, figure out which one is broken, and then returning the bad one.

wes-b, 2020-05-14:
Any change k4aviewer is generating errors?

ETLang, 2020-05-20:
Hi again. Sorry for the delayed response; I got sidetracked with some other tasks.

I swapped out my Kinects with two entirely new Kinects, and I'm seeing the same problem. No errors shown in the log, but the sub does repeatedly print the following warnings.

[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for imu
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth

wes-b, 2020-05-21:
Windows? For validating all this are you using only the cables provided to you with the Kinect? Besides the sync cable.  If you have USB Extension cables, please remove them.

Can you check to see what USB Host controllers your Kinects are using on Windows?

ETLang, 2020-05-21:
I'm using the cables they came with. I think the sync cable came with a stereo system I bought some years back. It carries audio just fine.

Here's an image of the USB host controllers and devices I *have*, but I don't know how to tell which the Kinects are actually using...

[image]

hagorms, 2020-05-22:
Hello. 
I currently work with 2 kinects and external signal gernerator. It runs without a problem on my pc. 
Do you have an oscilloscope to check the cable?  It would be great if you can send me sync out picture so I can compare it with mine. 
Best!

ETLang, 2020-05-22:
Sorry, I don't have an oscilloscope and I wouldn't know how to use one if I did! lol

ETLang, 2020-05-22:
In the vain hope that it might kind of work like an oscilliscope, I connected sync out on one Kinect to the audio line in port on my computer, and then tried recording it using Audacity. It successfully picks up the bit of static noise that results if I fiddle with the plug, so the setup in general is working. I figured maybe there's a chance it'd pick up on the signals being sent each frame (30Hz, audible in principle anyway).

Audacity picks up nothing. Maybe it wouldn't. I don't know. But that's about the best I can do!

ETLang, 2020-05-22:
Could I possibly hack together a signal generator by plugging Line Out into Sync In and writing code to play out a waveform as audio that would be interpreted as a signal?

hagorms, 2020-05-25:
@ETLang, I think the cheapest way is to buy arduino or rasberry py and use it as oscilloscope. Or just buy a cheap oscilloscope to test your electronic hardware.

wes-b, 2020-07-10:
Closing out this old issue, if you are still having problems we can reopen it.

closed 2020-07-10 by wes-b

arkader, 2021-10-31:
Hi @ETLang ,

Have you managed to fixed your issue with the sync?