microsoft/Azure-Kinect-Sensor-SDK #1215: SetCameraControl(). Invalid color control mode

furushchev, 2020-05-14:
**Describe the bug**

When opening a camera device and getting image data, an error occurs.

**To Reproduce**

1. Install libk4a and k4a-tools from apt-get and register udev rules
2. Connect a sensor to the computer
3. Run `k4aviewer` or `k4arecorder`
4. See an error

**Expected behavior**

Image can be fetched from a sensor

**Logs**

```
$ k4arecorder /tmp/output.mkv
Device serial number: 000261500312
Device version: Rel; C: 1.6.108; D: 1.6.79[6109.7]; A: 1.6.14
[2020-05-14 17:00:04.624] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1055): SetCameraControl(). Invalid color control mode

[2020-05-14 17:00:04.624] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (1019): color_set_control(device->color, command, mode, value) returned failure in k4a_device_set_color_control()
Runtime error: k4a_device_set_color_control() for auto gain failed 
[2020-05-14 17:00:04.891] [error] [t=18051] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 204.
[2020-05-14 17:00:04.891] [error] [t=18051] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2020-05-14 17:00:04.891] [error] [t=18051] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2020-05-14 17:00:04.891] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2020-05-14 17:00:04.891] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (398): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2020-05-14 17:00:04.892] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2020-05-14 17:00:04.892] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2020-05-14 17:00:04.892] [error] [t=18038] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (895): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
Runtime error: k4a_device_start_cameras(device, device_config) returned 1
```

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  Ubuntu 18.04
 - SDK Version:   1.4
 - Firmware version:  

```
$ AzureKinectFirmwareTool -q
 == Azure Kinect DK Firmware Tool == 
Device Serial Number: 000261500312
Current Firmware Versions:
  RGB camera firmware:      1.6.108
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft
```

**Additional context**

I see an error on `dmesg`:

```
[ 1946.063363] uvcvideo: Found UVC 1.00 device Azure Kinect 4K Camera (045e:097d)
[ 1946.064770] uvcvideo: Failed to set UVC probe control : -32 (exp. 26).
[ 1946.069810] uvcvideo 8-1.1:1.0: Entity type for entity Extension 2 was not initialized!
[ 1946.070003] uvcvideo 8-1.1:1.0: Entity type for entity Extension 6 was not initialized!
[ 1946.070133] uvcvideo 8-1.1:1.0: Entity type for entity Extension 5 was not initialized!
[ 1946.070267] uvcvideo 8-1.1:1.0: Entity type for entity Processing 3 was not initialized!
[ 1946.070396] uvcvideo 8-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 1946.079088] input: Azure Kinect 4K Camera as /devices/141a0000.pcie/pci0005:00/0005:00:00.0/0005:01:00.0/0005:02:04.0/0005:06:00.0/usb8/8-1/8-1.1/8-1.1:1.0/input/input11
[ 1971.622236] xhci_hcd 0005:06:00.0: WARN Event TRB for slot 4 ep 2 with no TDs queued?
[ 1971.876666] usb 8-1.1: usbfs: process 18050 (k4arecorder) did not claim interface 0 before use
```

- lsusb

```
$ lsusb
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 008 Device 010: ID 045e:097c Microsoft Corp. 
Bus 008 Device 011: ID 045e:097d Microsoft Corp. 
Bus 008 Device 002: ID 045e:097a Microsoft Corp. 
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 003: ID 045e:097e Microsoft Corp. 
Bus 007 Device 002: ID 045e:097b Microsoft Corp.
```

wes-b, 2020-05-14:
THANK YOU for the very detailed issue! 

Camera control is fixed with #1179 

The error about  `Depth engine create and initialize failed with error code: 204` sounds like you might be attempting to run over a remote connect / headless machine.

furushchev, 2020-05-15:
@wes-b Thank you for the comment!

> The error about Depth engine create and initialize failed with error code: 204 sounds like you might be attempting to run over a remote connect / headless machine.

Does it mean the driver node must be executed with display?

furushchev, 2020-05-15:
I connected to a display and tried to run the command again and found that the error now disappeared.
However, another error occurs on getting point cloud.

```
[2020-05-15 14:41:57.466] [error] [t=25994] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.
c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:  59796433 type:Depth 
[2020-05-15 14:41:57.663] [error] [t=25994] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.
c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:  59996355 type:Depth 
[2020-05-15 14:41:57.866] [error] [t=25994] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:  60196444 type:Depth 
[2020-05-15 14:41:58.061] [error] [t=25994] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.
c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:  60396355 type:Depth 
[2020-05-15 14:41:58.264] [error] [t=25994] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.
c (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:  60596455 type:Depth 
```

wes-b, 2020-05-15:
Good news on the progress! Lest start a new bug with these new errors. The key to getting you help will be a fresh set of repro steps for how you got here.

> Does it mean the driver node must be executed with display?

You do need an active user session for the depth engine to be able to use OpenGL

furushchev, 2020-05-18:
@wes-b 
> You do need an active user session for the depth engine to be able to use OpenGL

Got it. Thanks.

> The key to getting you help will be a fresh set of repro steps for how you got here.

I run `DISPLAY=:0 k4arecorder output.mkv` and got the error posted above.

furushchev, 2020-05-18:
Attached full logs again here:

```
$ k4arecorder output.mkv                                                                    Device serial number: 000261500312                                                                              
Device version: Rel; C: 1.6.108; D: 1.6.79[6109.7]; A: 1.6.14                                                   [2020-05-18 12:47:57.546] [error] [t=2571] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cp
p (1055): SetCameraControl(). Invalid color control mode                                                                                                                                                                        
[2020-05-18 12:47:57.546] [error] [t=2571] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (1019): color_set_control(device->color, command, mode, value) returned failure in k4a_device_set_color_control()              
Runtime error: k4a_device_set_color_control() for auto gain failed                                              [2020-05-18 12:48:02.416] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    200000 type:Depth  [2020-05-18 12:48:02.448] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    233344 type:Depth  
[2020-05-18 12:48:02.483] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    300222 type:Depth  
[2020-05-18 12:48:02.514] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    333511 type:Depth
[2020-05-18 12:48:02.548] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    366888 type:Depth
[2020-05-18 12:48:02.581] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    400211 type:Depth
...
```

wes-b, 2020-05-18:
Does K4AViewer run without these errors? Can you use K4aViewer to confirm the color camera is working as expected? The error means you are not getting color camera images.

furushchev, 2020-05-19:
@wes-b Hi,
I tried K4AViewer and got the same error after pressed "Start" button.

kalluwa, 2020-05-20:
hi, have you solved this problem?
any way to start the camera with  windows remote desktop?

furushchev, 2020-05-20:
@kalluwa Not yet.

> any way to start the camera with windows remote desktop?

It seems it is another problem. I recommend to open a new issue instead of posting here.

kalluwa, 2020-05-20:
@furushchev 
thanks

i noticed that:
>---Does it mean the driver node must be executed with display?
>You **do need** an active user session for the depth engine to be able to use OpenGL

window remote desktop cannot start opengl program normally( maybe opengl1.1 works but not for higher version)  just as your machine with no display

furushchev, 2020-05-25:
@kalluwa Sorry for the late reply and thank you for the advice.
I'm now trying to run the nodes with display and still getting the same error unfortunately...

@wes-b Any further advice?

Thank you very much in advance!

wes-b, 2020-05-26:
I am not sure why you are not getting Color images. Based on the log entry below, it happens ~5s into your recording. It might be you are having USB problems. You could try different USB ports. If you have extension cables you should remove those. Do you have another PC you could test with?

We see this on Windows when a bad driver has been installed on the PC or when images color images are leaked.

You might also try running at the lowest settings to see if it works on your PC at all. Recording takes a lot of resources to keep up with the data stream.

> [2020-05-18 12:48:02.416] [error] [t=2590] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/capturesync/capturesync.c
 (142): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    200000 type:Depth

furushchev, 2020-05-29:
@wes-b Thanks for the comment.

I finally found that this error occurs if the sensor device is connected to a computer on boot.
If it is connected to the computer after boot time, it works without the error.
I found that if the sensor is connected before the computer is booted, after running the command `udevadm trigger`, it now works.
Maybe the process on detection of the sensor is not completed if the sensor is connected to the computer before it is powered on...
Anyway, thank you very much for debugging together!

closed 2020-06-01 by furushchev