microsoft/Azure-Kinect-Sensor-SDK #1218: Question about temporal filtering of Kinect images

ArviSyrjanen, 2020-05-15:
Hello,

I was wondering if the incoming Kinect frames are truly raw individual frames, or is there some kind of temporal averaging happening under the hood?

I got the idea for this question from [here](https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera#systematic-error) as it is stated that

> We compute the temporal average over many frames of a static scene to eliminate depth noise as much as possible.

I understand that this question might seem a little bit silly, but the section in the documentation can be interpreted as if some temporal filtering is happening _all the time_ when operating the Kinect camera and so I was tempted to ask.

Thank you for reading this.

rabbitdaxi, 2020-05-18:
@ArviSyrjanen the depth processing in the SDK is per frame, we do not do temporal filtering. 

The documentation you are referring to is trying to use examples to define depth error (systematic and random error). The quote you had specifically describes how to **measure** depth jitter (random error) by computing the standard deviation after you capturing N depth frames.

ArviSyrjanen, 2020-05-18:
@rabbitdaxi thanks for the answer. Yes I was suspecting this was the case, but I was just curious. Aside from temporal filtering you probably don't do any other kind of filtering, hole filling etc. either on-chip or in the runtime API?

qm13, 2020-05-21:
@ArviSyrjanen depth processing is run on the host PC GPU. It makes use of filtering and other techniques to improve the depth image quality.

ArviSyrjanen, 2020-05-21:
@qm13 thanks for the answer.

Is it possible to retrieve raw depth frames e.g. via k4aviewer, the C++ API or ROS API, without the effect of filtering and other techniques?

qm13, 2020-06-15:
@ArviSyrjanen It is currently not possible to turn filtering off. A number of companies have requested this feature and we are investigating options. Why do you want access to the raw frames?

ArviSyrjanen, 2020-06-15:
> @ArviSyrjanen It is currently not possible to turn filtering off. A number of companies have requested this feature and we are investigating options. Why do you want access to the raw frames?

Thank you for your answer @qm13.
I am currently writing my masters thesis and a depth sensor evaluation is a part of it. I would like to analyse the Azure Kinect camera in a more decomposed way, try to reproduce error behaviour that is typical to CW ToF cameras and see how you guys have solved the problem(s) if they occur. Now I believe some behaviour is hidden from me in the post-processing layer.

To put it shortly, I am curious.

closed 2020-07-11 by qm13