microsoft/Azure-Kinect-Sensor-SDK #1221: Joints position and orientation not consistent if not facing the camera

TothiViseo, 2020-05-20:
Hello,

I am trying to use body tracking information in order to animate a 3D avatar. The result doesn't look so bad when the body faces the camera, however as soon as the body turns away from the camera, the orientations and coordinates do not match anymore.
It looks as though the orientations need to be negated around some axis.

[image]
[Body facing camera looks OK]

[image]
[Body turning back to camera seems inconsistent]

Is there a known reason for that behaviour?

What I do with the 3D avatar is that I use the Pelvis as a pivot and rotate all joints to compensate the pelvis rotation. 
Then I convert the orientations to my coordinate system and apply them to my skeleton.

I would expect that based on identical camera configuration, if I turn around from back to front and vice versa the animation based on positions and the 3D avatar based on orientations would keeping matching. 

For some reason they seem "inverted" when the body is not facing the camera.
Am I missing some factor I should take into account?

Tests were conducted on Windows 10, SDK 1.4.0, Firmware 1.6.102 rgb, 1.6.75 depth

Thank's.

qm13, 2020-06-04:
@TothiViseo what are you using to render your 3D avatar?

TothiViseo, 2020-06-04:
Hello @qm13,
I am using Babylonjs.

TothiViseo, 2020-06-09:
Hello @qm13,
I finally solved my problem. 
I ran the Unity example provided [here](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/sample_unity_bodytracking) and fed it with my own rotation data. 
I compared the two results and it turns out the specificity was on Babylonjs side. 
The wrists still seem off sometimes but the rotations seem consistent with the positions now.

closed 2020-06-09 by TothiViseo