microsoft/Azure-Kinect-Sensor-SDK #1223: Sync-out drift overtime

hagorms, 2020-05-22:
**Describe the bug**
Sync out drifts 

**To Reproduce**

1. Connect azure sync out signal generator to oscilloscope.
2. Connect master Signal Generator to oscilloscope
3. Compare phases over time.   
4. You will see the signal shift (phase difference change) overtime.

**Expected behavior**
   phase difference is stable

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.4.1
 - Firmware version:  <Current Firmware Versions:
  RGB camera firmware:      1.6.108
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

Device Serial Number: 000209301412
Current Firmware Versions:
  RGB camera firmware:      1.6.108
  Depth camera firmware:    1.6.79
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft>

**Additional context**
Sync out drifts  for frame rate 5, 15, 30

hagorms, 2020-06-08:
please look on the following screenshots.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Blue is azure sync out and yellow is master sync

hagorms, 2020-06-08:
The framerate is 5,
scale square 100ms,.
Blue = Azure Kinect.
Yellow Master Clock.

You can see that signal drifts. blue goes forward and meets yellow once again.

wes-b, 2020-06-08:
If I am reading things correctly:
* Blue is the output of your signal generator
* Yellow is the output of our sink out signal signal of a camera that is configured as a subordinate and had the signal generator plugged into the input.
* Timestamps in the image range from 258,015,622us to 830,415,622us resulting in a loss of 1 frame every ~ 14minutes at 5FPS.

hagorms, 2020-06-09:
Blue = independent Azure kinect master output sync.
Yellow = master Signal generator. It is independent. It sends only 5 pulses per second. 
I don't know if you can call it frame loss. If Azure generates exactly 5 frames per second, phase difference should be constant. Unfortunatelly, we see the blue (azure) drift.  And it meets the master clocke every ~14 minutes.

wes-b, 2020-06-16:
I was just informed by our team I misunderstood your setup. I thought the output of the signal generator was hooked up to the input of the subordinate camera. In this scenario we do not expect any phase shift.

What we understand your situation to be is that you are comparing the output of the Azure Kinect master device to the output of a signal generator. With the two signals not tied to each other in any way then the are *expected* to drift over time. The clock on your signal generator and the one in the Kinect will have ever so slightly different frequencies;  200.0001Hz vs 199.9999 Hz as an example.

hagorms, 2020-06-22:
Is it possible to be fixed? Does it have any relationship with  https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1227   ?

wes-b, 2020-06-22:
The 2 independent clocks are expected to drift over time.

closed 2020-06-22 by wes-b

jonahpearl, 2023-02-25:
Hey @hagorms did you ever resolve this? I am seeing a very similar drift (1 frame every 30 min at 30 Hz), despite sending in a us-precise sync (period 1 per 33,333 us) to two daisy chained azures. I will post a more detailed assessment soon as a new issue but wanted to see if you had figured out what was causing your drift, if it was consistent between cameras, etc. Thanks!

hagorms, 2023-05-09:
we didn't solve it. We just dropped it. It is a hardware issue I guess

jonahpearl, 2023-05-09:
Thanks. I did some more experiments / reading in the meantime and I think what wes-b was saying was that, you need to set the camera to be in "subordinate" mode in order for it to listen precisely to any incoming sync pulses. You can't have a camera in "master" mode and also pass in sync pulses. Similarly if you're not passing in sync pulses at all, there will definitely be drift. Take a look at the code [here](https://github.com/jonahpearl/pyk4a_multicam/tree/trigger_all_sub) if you'd like to see a working example for syncing the azures with other sources :)