microsoft/Azure-Kinect-Sensor-SDK #1224: Azure Master and Azure Slave devices have different frame numbers.

hagorms, 2020-05-25:
**Describe the bug**
I connected 2 Azure kinects according to the documentation.  After several recording I have different number of frames in master and slave devices.

How to allign frames with close-to-perfect synchronisation? 
For example master recording has 1501frames and Slave recording has 1497 frames. 
If you rerun the experiment, these numbers are different.  
Mostly the difference between master and slave is 1 or 2 frames. 
This issue was tested with 2 different kinects.
This issue was tested with --rate 5, 15, 30 

**To Reproduce**
Connect master sync out to slave sync in. 
Start slave recording, 
Start master recording. 
Compare frame numbers. 

**Expected behavior**
Best: Frame count in master and slave are equal and kinects are perfectly synchronized. 
Acceptable: Frame count difference is stable. So we can allign master frame X to slave frame Y for every recording. 

Unkown : I don't know how system will behave if we have more than 2 kinects. 

**Questions** 
Is there any way to synchronise kinect azure clock so we know when the frame was captured exactly. Is there any code sample how to allign kinect frames according to the system time, or how to get this system time in C/C++ interface

**Screenshots**

**Desktop:**
 - OS with Version: Windows 10. 
 - Firmware version : AzureKinectDK_Fw_1.6.108079014 
 - SDK 1.4.0

wes-b, 2020-05-27:
The device timestamp is synchronized acrossed all devices. Don't use frame counts.

Look at [get_synchronized_captures](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/3cfe362dacc4ad9b2206af3f28d62e1eb7be99c2/examples/green_screen/MultiDeviceCapturer.h#L95) in our green screen example for how to do it with a live stream.

For MKV's the device timestamps always start at zero and start_timestamp_offset_usec from k4a_record_configuration_t needs to be applied to get the recorded device timestamp. start_timestamp_offset_usec is effective an offset removed from every sample to they streams can start at zero. From there you can compare to the device timestamp from the other stream to create a synchronized pair.

hagorms, 2020-06-03:
Ok. I am working on it right now.

hagorms, 2020-06-08:
Here is my setup description when one kinect is a master : 
[image]
External camera starts recordning when it recieves a signal. 
Here are the screenshots from anoher recording with external camera. 

[image]

[image]

[image]
You can see the difference between RGB and IR image.

As we can see from screenshots: delay for both cameras is around 2 timestamp units

hagorms, 2020-06-08:
After I connected both kinect devices to external signal generator I get the following result 

[image]
[image]

For one kinect it is still 22 timestamp units, 
for another one it is 11 timestamp units.

hagorms, 2020-06-08:
Here are extra screenshots from Azure Kinect Viewer. (When running in slave configuration)
33 timestamp units difference
[image]
45 timestamp units
[image]

These numbers are not stable and have shift between 2 kinects.

wes-b, 2020-06-08:
Our device timestamps are in microseconds. We don't expect them to be exactly the same as they are from two different clocks on two different devices. We do expect them to be within 100us of each other, which from what I can tell is what you do see.

hagorms, 2020-06-09:
But why doesn't the difference between RGB and IR frame stays stable?

wes-b, 2020-06-12:
Can you be more specific about what you consider unstable? I think you are talking about 33us or 45us of difference between color and depth images.  This is real hardware with real clocks, and a calculation to find center of exposure.  You can't expect a perfect timestamp match.

closed 2020-07-10 by wes-b