microsoft/Azure-Kinect-Sensor-SDK #1225: Azure Master and Azure Slave frame shifts in the beginning

hagorms, 2020-05-25:
**Describe the bug**

Azure Master and Azure Slave recordings have non stable frame shifts. 
How to replicate: start mobile phone clock. Record mobile phone pictures with two devices. Try to align timer numbers in the beginning and in the end of the recording. Check frame  numbers and frame shifts between these measurements.

What you can get: 
- 00:00:30.10 is frame 10 on azure 1 and 00:00:30.10 on frame 7 on azure 2. 
- 00:00:30.10 in frame 10 on azure 1 and 00:00:30.10 on frame 8 on azure 2. 
- 00:00:30.07 in frame 10 on azure 1 and 00:00:30.10 on frame 9 on azure 2.
- 00:00:30.10 in frame 10 on azure 1 and 00:00:30.11 on frame 8 on azure 2.
All experiments can be found here: 
https://docs.google.com/spreadsheets/d/1mbifTFKxhFPOgZCIs_TITnPIxXd-gqtwXFSVJf_MkE8/edit?usp=sharing 

This issue was tested with 2 different kinects.
This issue was tested with --rate 5, 15, 30 

**To Reproduce**
1. Start mobile phone clock
2. Point Azure Master and Azure Slave to the mobile phone
3. start - stop several synchronized recordings. 
4. convert mkv 2 frames and compare timestamps. 

**Expected behavior**
Perfect: Master  and Slave kinect have perfectly alligned timestamps and frame numbers.

**Desktop :**
 - OS with Version: Windows 10
 - SDK Version:  1.4.0. 
 - Firmware version:  AzureKinectDK_Fw_1.6.108079014

wes-b, 2020-05-27:
This is by design, we make no attempt to align frame numbers. The device timestamp applied to each capture must be compared to find the synchronized images between 2 camera streams.

hagorms, 2020-06-03:
Yes, I got it. 
I currently look on the calibration example. It was noted that calibration example has functionality for frame synchronization based on device timestamps. I will upload this thread after I find a solution. 
I need to synchronize kinect with external device, that has only PC time. Do you have any ideas how can I do it?

wes-b, 2020-06-03:
The SDK also applies a system timestamp to image data when it arrives on the host PC. You can use this, but it is not as accurate as the HW timestamp applied by firmware. It is measured at applied with the frame is handled within the SDK. Depending on exposure this can be 10's of milliseconds later.

closed 2020-06-03 by wes-b