microsoft/Azure-Kinect-Sensor-SDK #1226: Azure Master and Azure slave frame drops

hagorms, 2020-05-25:
Azure master and azure slave devices have frame drops. 

**Describe the bug**

**To Reproduce**
1) start mobile phone clock.
2) Record mobile phone pictures with two devices.
3) Find two timer numbers on both cameras for example 00:00:30.10 and 00:05:30.10 .
4) Check frame count difference between timer values for Master and slave devices. 
5) repeat 1-4 several times. 

This issue was tested with 2 different kinects.
This issue was tested with --rate 5, 15, 30 
Firmware version : AzureKinectDK_Fw_1.6.108079014 

**Expected behavior**
Difference between timestamps for master and slave devices is always the same. 

**Link to Excell** 
https://docs.google.com/spreadsheets/d/1mbifTFKxhFPOgZCIs_TITnPIxXd-gqtwXFSVJf_MkE8/edit#gid=0 
Frame number differences are highlighted with red on columns RST

**Desktop**
 - OS with Version: Windows 10
 - SDK Version:  1.4.0
 - Firmware version:  AzureKinectDK_Fw_1.6.108079014

wes-b, 2020-05-27:
Dropped frames are to be expected. This can be due to PC performance and available USB Bandwidth. Memory and storage media performance play a part in recording speed as well.

hagorms, 2020-06-08:
If dropped frames are to be expected and in the beginning, frames are not synchronised, how do I  synchronise azure kinect with external device? 
According to github issues, system timestamp is also not the best solutution.

wes-b, 2020-06-08:
We expect the **device** time stamps to be within 100us of each other when running in master/subordinate mode.

[depth_delay_off_color_usec](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__device__configuration__t.html) and [subordinate_delay_off_master_usec](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__device__configuration__t.html) will need to be zero for the timestamps of the relative image to be within 100us of the master color image.

hagorms, 2020-06-09:
I understand that device timestamps are reliable units to synchronise between kinects. What I don't understand is how to synchronize external devices with kinect setup.
Lets imagine: a frame in the middle of recording is not recorded. We can find it by the timestamp. But external cameras recieved a pulse (pulses are stille generated) and recorded the dropped frame.
Another situation: kinect is a master and it sends pulses to record. Some frames in the beginning are not recorded by kinects, but pulses are generated. 
So it will be very hard to find the first external frame, which is synchronised with kinects.
Do you have any ideas how can I allign external frame sequences with kinects?

wes-b, 2020-06-12:
Checkout this comment: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/1240#issuecomment-643323350

closed 2020-07-10 by wes-b