microsoft/Azure-Kinect-Sensor-SDK #1227: Azure Devices generate pink and black frames in the slave mode with external master device.

hagorms, 2020-05-25:
How to replicate: 

**Describe the bug**

**To Reproduce**
1) Connect device to external signal generator.
2) Set external signal generator frequency to 5 , 15, or 30. 
3) Start Azure recording in slave mode.
4) Start signal generator.
5) After recording is finished convert mkv to frames with ffmpeg.

Some of the converted frames will be pink or black.
If 2 kinects were connected to Master device, pink frames are not synchronized. 
This issue was tested with 2 different kinects.
This issue was tested with --rate 5, 15, 30 

**Expected behavior**
No black or pink frames are recorded. 

**Desktop**
 - OS with Version: Windows 10. 
 - SDK Version: 1.4.0
 - Firmware version : AzureKinectDK_Fw_1.6.108079014 
**Additional context**
[image]
[image]

qm13, 2020-05-27:
How does the .mkv file look in the k4aviewer?

hagorms, 2020-06-03:
Dear qm13,
mkv files have some small black blinks in the k4aviewer. 
Do you mind have a look on the file?

wes-b, 2020-06-03:
Can you share a screen shot of K4AViewer and the black blinks you are referring to? We do expect a version of this to float across the depth camera, but it is not usually an entire frame.

hagorms, 2020-06-08:
Here are screenshots: 
[image]
[image]
[image]

wes-b, 2020-06-08:
Does this only reproduce with the external generator? Can you reproduce this with the Azure Kinect as the master?

hagorms, 2020-06-09:
It reproduces with external generator. If one of Kinect is a master, it might happen for one or two frames. After that it stays stable

wes-b, 2020-06-30:
The sync pulse width will cause this. Using a signal generator, you are very likely outside the tolerance of how long the pulse should be active. We are convinced yet that we want to fully document what the tolerance of this pulse needs to be to avoid this condition. But if you can record and emulate sync pulse generated by a master device for the color mode and resolution you will be using, then you should not have a problem going forward.

@markhog who also has questions in this area

closed 2020-06-30 by wes-b

tostercx, 2020-11-26:
Have a _very_ similar issue except the frames are red and black, not pink. Seems to be related to ambient temperatures getting lower, not sure why.