microsoft/Azure-Kinect-Sensor-SDK #1228: Non stable delay between sync in pulse and recorded frame.

hagorms, 2020-05-25:
**Describe the bug**
Azure Kinects may have a non stable delay between input pulse and  frame recorded. 

**To Reproduce**
1) Connect master signal to  2 azure devices in chain configuration. Master->Azure0->Azure1.
2) Point both Azure 1 and Azure 0 on a mobile phone with a running clock.
3) Record several sessions. 
4) Analyse session frames with ffmpeg. You may see different behaviour.

 You can get several options options.
Azure 0 recorded 00:01:28 .51 
Azure 1 recorded 00:01:28 .51

Azure 0 recorded a little bit earlier  00:01:28 .51 
Azure 1 recorded a little bit earlier  00:01:28 .53 

Azure 0 recorded a little bit later  00:01:28 .53 
Azure 1 recorded a little bit earlier  00:01:28 .51 

**Expected behavior**
Azure 0 and Azure 1 timestams are always the same. (repeat delay is not counted) 
Repeat delay is about 500 ns and shouldn't lead to 30 us difference between Azure 1 and Azure 0. 

**Oscilloscope test**
[image]
Master signal - yellow, Azure 0 Slave sync out - blue  

[image]
Master signal - yellow, Azure 1 Slave sync out - blue 

**Desktop (please complete the following information):**
 - OS with Version: Windows 10
 - SDK Version:  1.4.0
 - Firmware version:  AzureKinectDK_Fw_1.6.108079014 

**Additional context**
Table with several experiments : 

https://docs.google.com/spreadsheets/d/1mbifTFKxhFPOgZCIs_TITnPIxXd-gqtwXFSVJf_MkE8/edit?usp=sharing 
Different timestams are highlighted with pink.

wes-b, 2020-05-28:
The signal propagation delay from sync in to sync out of 300-400ns is expected.

If you sync camera frames based on device timestamp, instead of frame number, I expect you will find they align within 500us of each other.

hagorms, 2020-06-03:
Dear wes-b,
Yes. This behaviour you can see on the picture uploaded. The delayed signal has 300-400 delay. But unfortunatelly, Slave Kinect has delay of 20-30 us. If I connect Slave sync out to external camera (not Azure), I get my camera synchronized with master Kinect. Do you mind have a look on the mkv files?

wes-b, 2020-06-03:
400ns is tiny compared to our shortest exposure, which I believe is 500us IIRC. Exactly what issues are you running into?

BTW, what is this unit of time you are sharing: 00:01:28 .51? is .51 the same as 510ms? In this post I cannot tell where you see 20-30 us of delay so please try to explain that again.

closed 2020-07-11 by qm13