microsoft/Azure-Kinect-Sensor-SDK #1229: Realtime Body Tracking in C# doesn't work

tomassilny, 2020-05-26:
Hi,
I am trying to get body joints informations from Kinect Azure but it refresh every just 2 seconds and it is too slow, I need to get body joints informations every few milliseconds, is it possible? Please see my code below. I use `try` and `catch` it get body informations every 2 seconds and other time I get "Body Tracking Error" Exception.

```
using (Capture capture = await Task.Run(() => { return kinect.GetCapture(); }).ConfigureAwait(true))
{
     TimeSpan time = new TimeSpan(0, 0, 0, 0, 0);
          try
          {
               tracker.EnqueueCapture(capture, time);
               uint bodyCount = 0;
               using (var frame = await Task.Run(() => tracker.PopResult(time)))
               {
                    bodyCount = frame.NumberOfBodies;

                    if (bodyCount > 0)
                    {
                         mainFrame = frame;
                         DrawFaceFrameResults();

                    }
               }
          }catch(Exception ex)
          {
               Console.WriteLine("Body Tracking Error");
          }
  ...
```

I think here will be some problem with `EnqueueCapture()` function or `PopResult()` function.

I tried some solutions but still doesn't work. I also display RGB video from Kinect but this video works fine in realtime. 

Thank you.

qm13, 2020-05-27:
@tomassilny could you provide details of your host compute environment - CPU, RAM, GPU, VRAM?

tomassilny, 2020-05-27:
@qm13 thank you for your response.

CPU - Intel Core i5-8265U, 1.80 GHz
RAM - 8 GB
GPU - Intel UHD Graphics 620 4 GB
SSD storage (Dell)

We were using Kinect V2 and everything worked fine, but we would like to upgrade to Kinect Azure, we use this in the diagnosis of neurological diseases so that's why we need to track body joints in realtime.

Thank you.

qm13, 2020-06-04:
Currently we only support running on NVIDIA range of GPUs. Your application is running in CPU only mode hence the 1 fps frame rate. We are currently looking more performant DNN and other hardware acceleration options but have nothing to announce at the moment.

tomassilny, 2020-06-04:
@qm13 thank you for reply. And in the future do you expect support for other GPUs?

qm13, 2020-06-11:
The Body Tracking SDK uses the ONNX Runtime to execute models. We are working closely with our ONNX team in expanding support to other GPUs and expect to have an announcement regarding Windows support shortly.

closed 2020-06-15 by qm13