microsoft/Azure-Kinect-Sensor-SDK #1230: Memory leak when starting IMU

oilst, 2020-05-28:
When starting the IMU with                 
`_device.StartImu();`
the memory consumption increases steadily. As soon as I stop the imu with
`_device.StopImu();`
the memory consumption does not increase anymore. 

**Expected behavior**
Starting the Imu should not create a memory leak. 

**Desktop:**
 - OS with Version: Windows 1903 
 - SDK Version:  1.4.0
 - Firmware version:  
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14

qm13, 2020-06-01:
Could you please provide more precise repro steps? Is this C/C++ or C#?
Are you running one of the samples?
What tool are you using to monitor memory leakage?
Did you wait long enough for queue stabilization (wait beyond 15 seconds)?

oilst, 2020-06-02:
Steps to reproduce: 

-  checkout https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/csharp_3d_viewer
- add `device.StartImu();` after line 26 in Program.cs
- run the program in VS 2019 and monitor the memory in VS. The memory increases over time (I tried for 10 minutes and it is still increasing)

rapidengineering, 2020-09-11:
Same issue on my side

Vangos, 2020-10-06:
The issue is also occurring in C# and Unity3D. You can reproduce it easily:

### Unity3D
- Start any Kinect Unity application.
- Open the Profiler.
- Select Memory.
- Take a sample.

The `ManagedHeap.UsedSize` reaches 1 GB within a few seconds and keeps growing.

Also, you'll notice that the Task Manager's memory consumption is increasing.

breadmatt, 2021-05-14:
I discovered this issue too, with nuget package Microsoft.Azure.Kinect.Sensor 1.4.1. Are there going to be an update in the future?

gregorijk, 2021-09-28:
This is still an open issue. I've just spent couple of hours trying to find a memory leak in my code and it turned out to be this bug related to the IMU. :(