microsoft/Azure-Kinect-Sensor-SDK #1232: Body tracking sdk in Rsapberry Pi4

Michael-robo13, 2020-05-29:
I have a question about the body tracking sdk. Can it be used in raspberry pi 4? I want to obtain the skeleton data in system of raspberry pi4 + ros + azure kinect. Looking forward to your reply.

qm13, 2020-06-04:
@Michael-robo13 Neither the Sensor SDK nor the Body Tracking SDK currently work with Raspberry Pi. We have released the Sensor SDK on the NVIDIA Jetson range of products and are working on a version of the Body Tracking SDK targeting the Xavier variants. Longer term we are also looking at other hardware acceleration options.

closed 2020-06-11 by qm13

AswinkarthikeyenAK, 2020-09-02:
Hi @qm13 ,
is the body tracking sdk released for Jetson Nano? If could share some info on that it would be really great 

Thanks

qm13, 2020-09-14:
It will not be available on the Jetson Nano. We are working to make it available on the Jetson Xavier variants.

Azzedine-Touazi, 2020-09-25:
Please what about Jetson TX2. When are you expecting the body tracking SDK will be released for Jetson (TX2)? Thanks in advance.

qm13, 2020-09-26:
We will not prevent the Body Tracking SDK being used on a TX2. What frame rate you can achieve is unknown.

Azzedine-Touazi, 2020-09-26:
@qm13  Thanks again!

AswinkarthikeyenAK, 2020-10-29:
@qm13,
Any updates on when we might get the SDK Jetson Xavier variants? 
Thank you

Azzedine-Touazi, 2020-11-30:
@qm13,  
Please any update about body tracking SDK for Jetson.  

Thanks in advance.

Michael-robo13, 2020-11-30:
Sorry, I didn`t use the body tracking SDK for Jetson. I changed my plan.

Azzedine-Touazi, 2021-02-19:
Hi @qm13

Any update relative to this issue? If not, are there other AI devices which support Azure body tracking SDK? 

Cheers.

andrey-tsb, 2021-07-21:
Also looking forward for Kinect Azure support on embedded platforms like Rpi4 (not sure about Azure body tracking SDK development roadmap). Currently there are other solutions for body tracking on Raspberry Pi4, but only with other sensors.