microsoft/Azure-Kinect-Sensor-SDK #1234: How to write system timestamp into MKV file via k4arecorder in sync mode?

YinghaoHuang91, 2020-05-31:
Thanks for making and sharing this great tool! 

I wonder how to write and read the system time information into the recordings while in external sync mode. I think the default command to do this is:  `k4arecorder.exe --external-sync sub -e -8 -r 5 -l 10 sub1.mkv`

But I get 0 when I use `k4a_image_get_system_timestamp_nsec` to get the system timestamp for each capture. 

So can anyone tell me what I missed here?

wes-b, 2020-06-01:
@xthexder what is the tag that needs to be read to get the base timestamp?

xthexder, 2020-06-01:
System timestamps are currently not recorded in the MKV. If you don't need wall-clock timestamps, device timestamps should be sufficient to sync external devices.

If you need wall-clock time of the recording, you could write your own tag with the starting time. The timestamp of k4a_record_create() is included in the recording header, but unfortunately it's not currently available to read in the playback API.

Another limitation is the MATROSKA_TIMESCALE_NS define, which is currently hardcoded to only record timestamps in microsecond resolution. If you need more than this, you'll probably have to make a fork.

wes-b, 2020-06-02:
The device timestamps are our source of truth as they are captured by hardware and are a lot more accurate than the system timestamps.

YinghaoHuang91, 2020-06-03:
Thanks a lot for the answers. I also wonder how I can record data into mkv file if I use a third-party device as the trigger signal? Is this function currently supported in the k4arecorder.exe?

wes-b, 2020-06-03:
Just like you would use your external trigger with the Kinect in subordinate mode, I would expect the recording to work in the same way when put into subordinate mode.

```k4aRecorder <options> --external-sync subordinate```

closed 2020-06-03 by wes-b

DavidTorresOcana, 2024-05-03:
Feature implemented in #1978 by modifying the Matroska writer to set the `K4A_START_OFFSET_NS` flag as the first sample's timestamp from the device or the system, depending on user choice