microsoft/Azure-Kinect-Sensor-SDK #1235: Saving out a k4a_transformation_t ?

SeanStephensenAIRM, 2020-06-01:
I'd like to save out a k4a_transformation_t to a (probably csv) file so that it can be read back in later and assigned to a camera, but I can't seem to find the actual data structure of a k4a_transformation_t. [The docs](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/structk4a__transformation__t.html#details) don't seem to have any specific info on what is actually stored in this structure. Any tips on how to save this out?

qm13, 2020-06-04:
This is by design. Could developers please comment if they feel this is a necessary API. Please include a description of your use case.

SeanStephensenAIRM, 2020-06-04:
obviously this wouldn't be necessary for intrinsic transformations since they can be obtained so easily. I was hoping to use it to save extrinsic transformations so that re-calibration would not be needed if an experiment was interupted when using a stereo camera pair. For now, I've found a suitable solution by saving out the 4x4 transformation matrix (vector<vector<double>> format) which describes the extrinsic calibration, but a couple steps before the k4a_transformation_t (i.e. 4x4 matrix --> k4a_calibration_t --> k4a_transformation_t). This approach is fine, so I'm no longer worried about the more direct method, but k4a_transformation_t was my goal at first because it seemed like the most direct method. 

Feel free to close this issue as I have found a suitable workaround and have no compelling argument why a public k4a_transformation_t should be preferred.

closed 2020-07-11 by qm13

owen2t, 2022-11-30:
Just commenting to say that this would be very helpful for my application as well for the same reason. Thanks!