microsoft/Azure-Kinect-Sensor-SDK #1237: Can't capture color image.

ljhoo24, 2020-06-03:
Hello, everyone.
I'm trying to capture both color image and depth image from device.
but if I set the synchronized images only configuration to true, it wait infinitely to capture on k4a_device_get_capture function.
on the contrary to this, if I use synchronized images only configuration to false, It works. 
but k4a_device_get_capture function gives me only depth image handle.
color image handle have always nullptr.

for sure I already check on viewer on sdk folder and it had no problem to capture all images and other things(such as IMU/mic) on every configurations.

Environments - Windows 10 x64

wes-b, 2020-06-04:
Can you capture and share logs while you reproduce the issue/

ljhoo24, 2020-06-05:
Here is Log of my code below.

New logging session started

[2020-06-05 15:48:52.043] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (278): ******************** Device Info ********************
[2020-06-05 15:48:52.043] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (279): K4A SDK version:     1.3.0
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (285): Serial Number:       000035694412
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (289): RGB Sensor Version:  1.6.108
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (292): Depth Sensor Version:1.6.79
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (295): Mic Array Version:   1.6.14
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (298): Sensor Config:       6109.7
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (301): Build type:          Release
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (306): Signature type:      MSFT
[2020-06-05 15:48:52.069] [critical] [t=9136] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth\depth.c (312): ****************************************************
[2020-06-05 15:48:53.456] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    266888 type:Depth
[2020-06-05 15:48:53.490] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    300222 type:Depth
[2020-06-05 15:48:53.522] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    333522 type:Depth
[2020-06-05 15:48:53.556] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    366888 type:Depth
[2020-06-05 15:48:53.589] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    400211 type:Depth
[2020-06-05 15:48:53.623] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    433555 type:Depth
[2020-06-05 15:48:53.657] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    466877 type:Depth
[2020-06-05 15:48:53.691] [error] [t=21780] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\capturesync\capturesync.c (144): replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    500222 type:Depth

I just copied little bit of my log.
but the same error code is  acquired infinitely and repeatedly.
configuration was synchronized_images_only = false.

Strangely, when I run the viewer in sdk, this error code is not printed.

wes-b, 2020-06-05:
Does the color camera work with the Windows Camera app?

With no error message from the color camera there is no clear direction to investigate. The system log in the Viewer System might have a Media Foundation error, or an error from another component logged to help us. You might have other camera software installed causing a problem.

Can you describe in more detail what is happening in K4AViewer?
Does anything in #459 help?

ljhoo24, 2020-06-08:
Yes, color camera is working with windows camera app.
and I didn't install any other camera software on my pc.

about k4aviewer, it gives me a log only about device information such as sdk versions not an errors.
but it's definitely works !
It shows me a color images and depth image with any configurations.

more weird thing is that I can record color image and read image with SDK and my device !
Only impossible thing is that I can't acquire color image in real time with SDK.

I already checked #459 and that issue is not related with my problem.

> Does the color camera work with the Windows Camera app?
>
> With no error message from the color camera there is no clear direction to investigate. The system log in the Viewer System might have a [...]

ljhoo24, 2020-06-08:
additional information.
if I connect my camera device to other desktop, I can acquired color image handle from device.

I attached pc information of error acquired one below.

cpu - intel i7-6700
ram - 16gb
gpu - geforce gtx 980 ti

wes-b, 2020-06-08:
It sounds like our tools are working fine on your PC, but your code is not. Correct?

Ensure you are calling [k4a_image_release](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga27c81863b13fafc3934a32935a014e9f.html#ga27c81863b13fafc3934a32935a014e9f) and [k4a_capture_release](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga0ed6f74ec403c3eac1b8ef3afb32cee6.html#ga0ed6f74ec403c3eac1b8ef3afb32cee6) as appropriate. If you are leaking the color image on Windows, Media Foundation will silently stop providing images when it runs out of resources.

ljhoo24, 2020-06-09:
yes, you are right.

perfectly working on my pc using k4aviewer. but not my code.
strange thing is that I can run and get a color image handle in same code at different pc.
even that pc is worse than mine.

besides, after reading your comments, I changed my code and test it again.
but releasing handle is not working.
I mean releasing itself is worked, but still my code cannot get a color image handle on my pc.

same again, changed code can capture color image handle at different pc.

can you explain more details about out of resources ?
what resources should be required exactly.

wes-b, 2020-06-12:
I don't have any specifics unfortunately, we know Media Foundation has a limited number of capture buffers it preallocates. When they are used up Media Foundation stops delivering more images until they are returned. As a user of K4A you return them by releasing your captures and images.

As long as you call k4a_capture_release() on all your captures, and k4a_image_release() for all your images, you should not run into this problem.

ljhoo24, 2020-06-16:
OK. I know that my problem is strange situation.
and nobody can't help me to solve this.
issue will be closed now.

closed 2020-06-16 by ljhoo24