microsoft/Azure-Kinect-Sensor-SDK #1238: Misalignment in Pointcloud

rapidengineering, 2020-06-03:
Hello all,

Information:
-C# WPF
-Visual Studio 2019 16.6.0
-Windows 10 Professional
-Microsoft.Azure.Kinect.Sensor 1.4.0 (NuGet package)

            Sensor.StartCameras(new DeviceConfiguration()
            {
                ColorFormat = ImageFormat.ColorBGRA32,
                CameraFPS = FPS.FPS30,
                ColorResolution = ColorResolution.R720p,
                DepthMode = DepthMode.NFOV_2x2Binned,
                WiredSyncMode = WiredSyncMode.Standalone,
                SynchronizedImagesOnly = true
            });

I have two streams:
- Left stream is 2d Image stream with resolution 320x288
- Right stream is Point Cloud stream in 3D Viewport.

We detect an object in left stream and draw a boundingbox around the detection, but we cant get the boundingbox to allign with the object in the pointcloud stream. 

_Left stream:_

                    Image imagecol = Transformation.ColorImageToDepthCamera(cap);
                    BGRA[] colorPixels = imagecol.GetPixels<BGRA>().ToArray();
              
                    var region = new Int32Rect(0, 0, ResolutionWidth, ResolutionHeight);
                    Writeablebitmap.Lock();
                    unsafe
                    {
                        using (var pin = imagecol.Memory.Pin())
                        {
                            Writeablebitmap.WritePixels(region, (IntPtr)pin.Pointer, (int)imagecol.Size, imagecol.StrideBytes);
                        }
                    }
                    Writeablebitmap.Unlock();
`

_Right stream (3d):_

`
                    
                    Image depthimage = Transformation.DepthImageToPointCloud(cap.Depth);
                    Short3[] xyzArray = depthimage.GetPixels<Short3>().ToArray();

                    for (int i = 0; i < num; i++)
                        {
                            veccol.Add(new Vector3(xyzArray[i].X, xyzArray[i].Y, xyzArray[i].Z));
                            colcol.Add(new Color4(colorPixels[i].R / 255f, colorPixels[i].G / 255f, colorPixels[i].B / 
                                 255f, colorPixels[i].A / 255f));
                        }
                  PointModel.Positions = veccol;
                  PointModel.Indices = intcol;
                  PointModel.Colors = colcol;

`

_This is how we scale the boundingbox to the pointcloud stream:_

We know the business card dimensions are 85x55 (mm). 
To scale the boundingbox we divide the card dimensions with the pixeldimensions of the detection.
` 
         scaleX = 85 / bbox.Width;
                scaleY = 55 / bbox.Height;
`
(Note, We only use this once for calibration )

_this is how we transform the boundingbox to the pointcloud stream:_

```
                                var mesh = new MeshBuilder();

                                //transform boundingbox
                                double x_box = (oldbox.Left - (double)ResolutionWidth / 2) * scaleX;
                                double y_box = (oldbox.Top - (double)ResolutionHeight / 2) * scaleY;
                                double width = oldbox.Width * scaleX;
                                double heigth = oldbox.Height * scaleY;
                                var newbox = new Rect(x_box, y_box, width, heigth);

                                //draw boundingbox
                                mesh.AddBoundingBox(new System.Windows.Media.Media3D.Rect3D(newbox.X, newbox.Y, ZLimit, newbox.Width, newbox.Height, 1), 2);
```

Screenshot of the two Streams: (Note: the middle card is exactly in the middle of the pointcloud)

[image]

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

rapidengineering, 2020-06-09:
Essentially what I want to achieve, is to perform a ColorImageToDepthCamera transformation without getting the Fisheye-ish effect on the color stream.

qm13, 2020-06-12:
We believe this is a hardware issue. Can you please report the issue [here](We believe this is a hardware issue. Can you please report the issue here)))

closed 2020-07-11 by qm13