microsoft/Azure-Kinect-Sensor-SDK #1242: Azure Kinect Body tracking SDK crashes on opening

hpsinghjk, 2020-06-04:
[image]
[image]

**Describe the bug**
Azure Kinect Body tracking SDK crashes on opening

**To Reproduce**
 build 18362.836
 - SDK Version:

hpsinghjk, 2020-06-04:
Hello, I wanted to perform body tracking for shoulder movement. The Azure kinect viewer opens up on the computer and I can see the depth camera information but the kinect body tracking does not open and crashes as soon as I click on it. I have tried uninstall and reinstall but it is showing a different icon compared to the viewer as can be seen in the picture above. Please advise what other information would you require.

closed 2020-06-05 by hpsinghjk

lisajamhoury, 2020-10-14:
@hpsinghjk Did you solve this issue? Having the same problem and cannot find a solution.

lisajamhoury, 2020-10-14:
We fixed this with the firmware update. https://docs.microsoft.com/en-us/azure/kinect-dk/update-device-firmware