microsoft/Azure-Kinect-Sensor-SDK #1243: libtorch libraries trigger  a bug

bigpo, 2020-06-04:
**Describe the bug**

when I linked the libtorch libraries in CMakeLists.txt file like this:
`target_link_libraries( index_map ${TORCH_LIBRARIES})`, I got an error to run the code.
code comes from `https://github.com/UnaNancyOwen/AzureKinectSample/tree/master/sample/cpp/index_map`
**To Reproduce**

1. It's OK without link the libtorch.
2. err occurs if link the libtorch and run the `index map` . like `Signal: SIGSEGV (Segmentation fault)`
3. err in line `k4a_result_t result = k4abt_tracker_create(&sensor_calibration, config, &handle);` file `/usr/include/k4abt.hpp`.

**Expected behavior**

no err
**Logs**

```
libusb: warning [handle_timeout] async cancel failed -5 errno=22
[2020-06-04 19:29:45.126] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2020-06-04 19:29:49.196] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2020-06-04 19:29:51.532] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_depth" was popped in a disabled state.
[2020-06-04 19:29:51.996] [error] [t=23050] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_color" was popped in a disabled state.
libusb: warning [handle_timeout] async cancel failed -5 errno=22
[2020-06-04 19:29:58.436] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color/uvc_camerareader.cpp (1300): DecodeMJPEGtoBGRA32(). MJPEG decode failed: -1
[2020-06-04 19:30:09.582] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_depth" was popped in a disabled state.
[2020-06-04 19:30:09.593] [error] [t=23050] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_color" was popped in a disabled state.
[2020-06-04 19:30:09.598] [error] [t=23073] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/queue/queue.c (112): queue_pop(). Queue "Queue_color" was popped in a disabled state.
```
**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: ubuntu 18.04
 - SDK Version:  libk4abt1.0 
 - Firmware version:  

**Additional context**

qm13, 2020-06-04:
What version of the SDKs are you using. There is not enough information for us to provide definitive guidance. We suspect that the JPEG is corrupted either because the CPU cannot keep up or the memory has been overwritten.

bigpo, 2020-06-04:
> What version of the SDKs are you using. There is not enough information for us to provide definitive guidance. We suspect that the JPEG is corrupted either because the CPU cannot keep up or the memory has been overwritten.

Thanks for your reply. I will supplement the info about SDKs after I back to work. I was curious why the libtorch could trigger the bug?

closed 2020-07-10 by qm13