microsoft/Azure-Kinect-Sensor-SDK #1245: Kinect data on GPU with `gpu::GpuMat`, `ogl::Buffer`, `ogl::Texture2D`, or OpenGL handles

diablodale, 2020-06-07:
Add access to Kinect data via `cv::gpu::GpuMat`, `ogl::Buffer`, `cv::ogl::Texture2D`, and/or OpenGL handles to keep data on GPU and greatly reduce GPU<->CPU memory churn for image processing, computation, etc. As I read the code today, almost all work by this SDK codebase is done on the CPU. (depthengine is not part of this SDK codebase)

For the same reasons that depthengine is written for GPU usage...our apps need to use the GPU. Developers on tiny industrial platforms up to workstations need the Kinect data to stay on the GPU to scale across the variety of computing platforms. Forcing download of all frames to CPU is outdated 👴 and wastes resources of gpu, cpu, power, and thermal budget.

## Examples

OpenCV provides `cv::gpu::GpuMat` as one container for data kept in GPU memory. Two others are `ogl::Buffer` and `cv::ogl::Texture2D`. If depthengine via this SDK exposed Kinect data in such containers, it would allow easy usage in OpenCV and libraries using OpenCV interfaces. In addition, data can be downloaded GPU->CPU at any time with single lines of code `cv::gpu::GpuMat::download()`.

```c++
auto image = k4a_capture_get_color_image(capture);
cv::gpu::GpuMat sourceOnGpu = k4a_image_get_gpumat(image);
cv::gpu::GpuMat resultOnGpu;
cv::gpu::threshold(sourceOnGpu, resultOnGpu, 128.0, 255.0, CV_THRESH_BINARY);

cv::Mat resultSlowCpuMem;
resultOnGpu.download(resultSlowCpuMem);  // <-- only here do we request data move from GPU->CPU
```

Or imagine depthengine exposes the data in a capture collection of OpenGL handles to shared textures, pbo, etc. And enabling dirextx/opengl texture sharing as needed for depthengine. This would allow the following pseudocode. Note:  in this pseudocode *zero* image data was copied from GPU->CPU memory.

```c++
auto capture = k4a_get_capture();
auto image = k4a_capture_get_color_image(capture);
auto glTexture = k4a_image_get_buffer(image);
myShader.use();
myShader.setInt("texture1", glTexture);
myShader.run();
```

## Savings

The memory savings, cpu savings, and cache churn is gigantic. For example, every NV12 4k frame is approximately 12MB of data. At 30fps that is approximately 365 MB every second and invalidating large chunks of the L1-3 CPU cache. And worse...in today's apps that must process data with GPUs this number is *doubled*. Why? Because depthengine forces download to CPU and the very next line of code is an upload of that same data back to the GPU...

```c
cv::gpu::GpuMat churnMemory(const k4a_image_t& image) {
  cv::gpu::GpuMat gpuMem;
  uint8_t * cpuMem = k4a_image_get_buffer(image); // download gpu->cpu
  gpuMem.upload(cpuMem); // duplicate resource waste cpu->gpu
  return gpuMem;
}
```

By keeping data on the GPU, **730MB every second is reduced to zero**. Plus, it allows for efficient GPU processing of data when often the destination is the GPU.

## Alternative

Currently none possible. All image data is forced to be copied to CPU memory. This SDK only exposes Kinect capture data via CPU memory using `k4a_capture_get_depth_image()`, `k4a_capture_get_color_image()`, or `k4a_capture_get_ir_image()`. Even the SDK examples do expensive GPU->CPU copies on every frame

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/aaddc09296731905ee775a1da30b95e06f7f6b97/examples/viewer/opengl/texture.cpp#L69-L77

qm13, 2020-06-08:
This is an excellent proposal @diablodale. I have also posted it on User Voice here:

https://feedback.azure.com/forums/920053-azure-kinect-dk/suggestions/40619830-kinect-data-on-gpu-with-gpu-gpumat-ogl-buffe

We would like to here from the community.

nvibd, 2020-10-07:
:heavy_plus_sign: 

Also running into this limitation, preventing better performance.

I'd really prefer a `k4a` device image type over an OpenCV one, though. Using an OpenCV type introduces a huge dependency, as well as additional licensing bureaucracy.

IMHO, a good location for introducing this is as early as needed to avoid any device->host copies. Maybe even in the device handle configuration.

michalsuchanek, 2022-07-07:
As outlined above, it would make an enormous difference is k4a_image_t exposed a GL texture handle. Any developments on this front after 2+ years? Loss for words...