microsoft/Azure-Kinect-Sensor-SDK #1246: Missing camera performance evaluation result

k-maheshkumar, 2020-06-08:
**Describe the bug**
I can see different metrics explained for evaluating the camera's performance on the official documentation page. But the result for the same for the different working ranges has not been published.

**Describe the solution you'd like**
Some publicly accessible (and citeable) metrics indicating how accurate the camera is.

**Additional context**
https://docs.microsoft.com/en-us/azure/kinect-dk/depth-camera

wes-b, 2020-06-08:
I think you are after this: https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification

k-maheshkumar, 2020-06-09:
No, I am looking for metrics results such as z accuracy, repeatability, plane fit RMSE, Peak Signal to Noise Ratio (PSNR)

wes-b, 2020-06-12:
Z accuracy is in the link I shared: https://docs.microsoft.com/en-us/azure/Kinect-dk/hardware-specification#depth-camera-supported-operating-modes.

closed 2020-06-24 by wes-b