microsoft/Azure-Kinect-Sensor-SDK #1247: how to set the stride_bytes argument in the k4a_image_create() for the mjpg

GitZinger, 2020-06-09:
**Describe the bug**

I tried to put 0 or 1 as the stride_bytes in there but got the error with "mjpg does not have a constant stride"? 
what should I do? 
please help me here! thank you so much!

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version:  1700
 - SDK Version:   1.2.0
 - Firmware version:  

**Additional context**

qm13, 2020-06-11:
There is no stride for mjpg. What issue are you encountering?

closed 2020-06-24 by wes-b

GitZinger, 2020-06-24:
then
1. I mean how can I declare a handler variable from the k4a_image_create()  to  hold a mjpg color image?
2.  then how to save the mjpg image into file? 
thanks @qm13

wes-b, 2020-06-25:
Use k4a_image_create_from_buffer() and set the stride to 0.

GitZinger, 2020-06-26:
@wes-b 
that is why i got this issue:
I tried to put 0 or 1 as the stride_bytes in there but got the error with "mjpg does not have a constant stride"?

wes-b, 2020-06-26:
I expect you would get this error from k4a_image_create(), but not k4a_image_create_from_buffer(). Also you are on an old SDK version so please update and retest with k4a_image_create_from_buffer().