microsoft/Azure-Kinect-Sensor-SDK #1251: Depth stream not working

pajama, 2020-06-11:
In AzureKinectViewer after starting the cameras with the depth camera enabled I get:
```
[ warning  ] : usb_cmd_libusb_cb(). USB timeout on streaming endpoint for depth
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    
```

- This only happens with one of my devices. My other two azure kinects are fine. 

- I've tried swapping out cables with no luck. 

- I've also tried the device manager suggestions from issue #459 

- The device drivers show up fine (with no warning indicators) in device manager. 

- The color stream works fine.

Thank you for your help!

qm13, 2020-06-11:
@pajama are you saying that the bad Azure Kinect does not stream depth and IR i.e. you see no IR or depth image in the k4aviewer. Please post a screen shot.

pajama, 2020-06-11:
correct. no depth/ir stream. 
[image]

However, other streams show fine if I disable the depth camera.
[image]

wes-b, 2020-06-12:
Looks defective, please exchange it.

You could attempt to factory reset and reprogram the device firmware.

pajama, 2020-06-12:
@wes-b can you get me in touch with someone for an exchange? I spent 2 hours on the phone today getting passed around randomly by Microsoft support trying to do just that. 

Thank you!

wes-b, 2020-06-12:
@qm13 should be able to point you in the right direction

pajama, 2020-06-13:
@qm13 I'm filming an event with the Kinect on the 22nd. Should I order a replacement assuming you'll be able to help me return the defective unit separately? Thanks so much!

qm13, 2020-06-15:
@pajama to meet your filming deadline I suggest you order a replacement. You can request a replacement [here](https://support.microsoft.com/en-us/supportforbusiness/productselection?sapId=f77b1b95-721e-43a0-2db8-b01e81a3f813)

closed 2020-07-11 by qm13