microsoft/Azure-Kinect-Sensor-SDK #1254: Color camera driver suddenly doesn't work anymore

ShikiGami, 2020-06-16:
I was able to use the Kinect device a few months ago, and now I tried on the same PC with same pc and it would open.

After looking a little bit, it seems that the color camera driver somehow isn't working.
It tells me to "Reinstall Driver", but if I try to reinstall it it gives me an driver installation error.

The only big difference between now and when I was able to use the device is that I updated Windows 10 from 1909 to 2004

[image]

[image]

[image]

[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\mfcamerareader.cpp:214 ) : RuntimeClassInitialize(). Failed to find color camera in open camera: 0xc00d36d5
[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\color\color.cpp:73 ) : SUCCEEDED( Microsoft::WRL::MakeAndInitialize<CMFCameraReader>(&color->m_spCameraReader, (GUID *)container_id)) returned failure in color_create()
[ error    ] ( D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c:177 ) : color_create( device->tick_handle, container_id, serial_number, color_capture_ready, handle, &device->color) returned failure in k4a_device_open()

wes-b, 2020-06-16:
Can you right click on the device with the yellow triangle and select "Uninstall device"? If an option appears to remove or delete the driver then choose that too. If that succeeds you can then detach and attach the camera to trigger the driver reinstall or in Device Manager you can select Action -> Scan for hardware changes.

ShikiGami, 2020-06-16:
> Can you right click on the device with the yellow triangle and select "Uninstall device"? If an option appears to remove or delete the driver then choose that too. If that succeeds you can then [...]

I can uninstall or remove the device, but after I re-plug the device, or try to scan for hardware changes it goes back to the icon with the warning sign.

wes-b, 2020-06-16:
I just noticed from your first image you are connected to an AMD USB 3.1 Extensible Host Controller. Do you have any Intel, TI, or Renasus Extensible Host Controllers on your machine? If you do you will need to use one of the other USB ports on your PC. If you don't then I would suggest purchasing on USB controller with one of these 3 chipsets.

I know you said this worked before an update. Perhaps it worked on another USB port (with another host controller) before the update?

ShikiGami, 2020-06-18:
I connected the device with the cabe I left there the last time I used it, so it is the exact same port.
Any incompatibility issue was probably introduced by the Windows version update or some other Windows update.

wes-b, 2020-06-24:
Any better luck with other USB ports?

ShikiGami, 2020-06-25:
> Any better luck with other USB ports?

Tried every single port of the PC, and it is exactly the same.
Just for a little more information, I'm using an X570 AORUS PRO motherboard, using a Ryzen 9 3900X CPU.

ShikiGami, 2020-06-25:
Just a quick update.
It seems that when you get the `One of the installers for this device cannot perform the installation at this time.` driver installation error, it means that you Windows installation is corrupted, and you are not longer able to install new drivers.
I tested it trying to install new drivers for my GPU, and sure enough, I wasn't able to do it.

I reinstalled Windows 10, and now I'm able to make use of the Azure Kinect DK.

It seems this is a bug/problem from Windows 10, and not so much a problem from the Kinect device or the SDK drivers.

wes-b, 2020-07-10:
Thanks for the update!

closed 2020-07-10 by wes-b