microsoft/Azure-Kinect-Sensor-SDK #1255: Secondary color image in the reference frame of primary color image?

SeanStephensenAIRM, 2020-06-16:
I'm using a stereo camera pair and code that is roughly based on the greenscreen example. I've been able to write out the secondary camera's depth image transformed into the perspective of the primary color camera using:

`successful = k4a_transformation_depth_image_to_color_camera(k4a_transformation_t seconday_depth_to_main_color, k4a_image_t secondary_depth_image, k4a_image_t transformed_image);`

shown here is the primary depth image in the primary color reference, the untransformed secondary depth image (secondary color reference), and the transformed secondary depth image (primary color reference frame):

[image]

[image]

[image]

I'd like to get a similar result to the last image, but where the secondary color image is shown in the primary color's reference, but the only k4a functions seem to be depth--> color or color--> depth. Is there any way to accomplish this with k4a functions? or would this OpenCV territory?

qm13, 2020-06-18:
When you refer to stereo cameras do you mean two Azure Kinect cameras? The greenscreen samples demonstrates calibrating two Azure Kinect cameras and then transforming the depth image from one of the cameras to the other camera.

SeanStephensenAIRM, 2020-06-18:
@qm13 that's correct, we have a stereo pair of kinects. I'd like to view the RGB images from the 2 kinects in a common reference frame using the calibration obtained from the chessboard calibration

closed 2020-07-11 by qm13