microsoft/Azure-Kinect-Sensor-SDK #1256: Error E1696 cannot open source file "k4a/k4a.hpp" | green screen example

jessekirbs, 2020-06-17:
When trying to build ALL_BUILD in the green screen project within Visual Studio 2019, I get the following error:

`Error (active)	E1696	cannot open source file "k4a/k4a.hpp"`

I've tried:
- Installing the Kinect Azure libraries via NuGet
- Including a k4a folder in the project root with k4a.hpp inside, 
- Right clicking _ALL_BUILD → Properties → Configuration Properties → VC++ Directories_ and adding the path to k4a.hpp under _Include Directories_.

**To Reproduce**
1. Use CMake GUI to configure and generate project files.
2. Open Project.sln
3. Right click ALL_BUILD in Solution Explorer
4. Click Build
5. Error appears in Error List

**Desktop (please complete the following information):**
 - Windows 10 Version 1909 for x64
 - Azure Kinect SDK v1.4.0

jessekirbs, 2020-06-26:
I was able to solve this issue by: 

1. Using git to clone the repository with:

`git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git`

2. Copying 'depthengine_2_0.dll' from the path _…\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\bin_ to my self-compiled 'bin' folder.

3. Move the contents of the _...bin\Debug_ folder into the _bin_ folder

4. Run executable.

closed 2020-07-14 by wes-b

srinivas-varadharajan, 2020-09-02:
@jessekirbs I'm running into the same issue. What do you mean by self-compiled 'bin' folder in the above comment?

jessekirbs, 2020-09-09:
@srinivas-varadharajan The bin folder that was created after running Generate in CMAKE. So:

1. Use git to clone the repository with:
`git clone https://github.com/microsoft/Azure-Kinect-Sensor-SDK.git`

2. Run CMAKE and generate project files.

3. Open K4A.sln in Visual Studio

4. Build all.

5. Copy 'depthengine_2_0.dll' from the path _…\Azure Kinect SDK v1.4.0\sdk\windows-desktop\amd64\release\bin_ to the _'bin'_ folder generated from CMAKE.

6. Move all files from the CMAKE location _...bin\Debug_ folder up one level to the _'bin'_ directory.

7. Run chosen .exe

Let me know if that makes sense and if it works for you. Otherwise, I can try to help.